devbis/z03mmc #90: Zibgee device never appear in Z2M devices

Neonox31, 2023-12-10:
Hello 👋

First, thanks for this awesome repo !
I bought 4 LYWSD03MMC few yers ago and yesterday I started to flash them using the zigbee firmware.
I carefully followed the OTA flash procedure and it the flash went successful for all devices 👍

The first 3 flashed devices are correctly detected by Z2M but the last one is never detected.

**What I tried :**
- Change the battery by a new brand one
- Remove the battery and wait few seconds to reset
- Strap ground to reset pin for few seconds but did no any effects on the screen, is this normal ?
- Flash directly the zigbee firmware using USB and a 1kΩ resistor : flash succeed and screen working but no device into Z2M
- Flash the OTA firmware using USB and a 1kΩ resistor and use OTA method to reflash the zigbee firmware, all went well but again no device into Z2M

**My setup USB flashing :**
[image]

**Used versions :**
ATC_ota_40000.bin -> 1.0.6
z03mmc.bin -> 1.0.6
Z2M -> 1.34.0

Maybe I missed something but now I'm out of ideas, if someone here could help me please ?

Thanks.

devbis, 2023-12-10:
Some users reported that after correct flashing it still remains with the previous firmware. So several (10 was reported) attempts may require.
I suggest follow this steps:

0. Use python script, don't use web fasher as it doesn't verify the flashing procedure is actually running
1. Use 'ea' command to erase all flash. It shouldn't display anything on the display for this time. If it still works, repeat ea command
2. Write firmware again until display starts displaying values again. 

Now you should be with zigbee fw, enjoy. :-)

Neonox31, 2023-12-10:
@devbis Thanks for your response, I will try to follow your recommendations and will keep this issue updated.

closed 2023-12-29 by devbis