devbis/z03mmc #93: Calibration / offset

bipsendk, 2023-12-14:
Is there some way to configure an offset or calibrate a flashed device ?

I have a device that seems to always show a humidity 9% higher than other references.
I know, that I can put an offset value into Z2M - but this only helps on values exposed to e.g. HomeAssistant via MQTT .. It would be nice if the display value of the device actually also was correct..

pvvx, 2023-12-17:
Today this is only possible for [another beta firmware version](https://github.com/pvvx/ZigbeeTLc/issues/30). 
The same is true for the option with simultaneous operation of [Zigbee and BLE](https://github.com/pvvx/BZdevice).

devbis, 2024-03-15:
It is supported starting from version [1.1.0](https://github.com/devbis/z03mmc/releases/tag/1.1.0).

closed 2024-03-15 by devbis