devbis/z03mmc #94: HV B1.4 + ATC_ota_40000.bin = brick

Regressor, 2023-12-15:
I flashed two devices with ATC_ota_40000.bin and got two bricks. No screen, no BT. 

I soldered CH340 usb to ttl adapter (3,3V) but allways  getting "Use reset chip (RTS-RST): see option --tact". I tried to change speed down to 340000 and only -b 480000 sometimes shows 1 or 3 sectors flashing. Changing timeouts up to 0.5 didn't help...

Connection: rts to vcc, rx to P14, tx to 1K then to P14, gnd to gnd.

Regressor, 2023-12-15:
Flashed with https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
python version unusable :(

devbis, 2023-12-15:
Thanks for the report. New transitional firmware from yesterday was not thoroughly tested. And now it is tested and good to go. Sorry for inconvenience.

closed 2023-12-15 by devbis

Regressor, 2023-12-27:
Got one more sensor. Flashed it over ait. Fix confirmed.