devbis/z03mmc #95: Отваливание термометров из сети Zigbee.

pvvx, 2023-12-16:
@devbis - замеченные проблемы:

Если не некий период отключить координатор, тогда устройства переключаются на роутеры.
После переключения на роутеры “умных розеток” Tuya, к термометрам не проходят ни какие Zigbee команды. 
Через неопределенный срок прерывается и передача показаний от термометров.
При уменьшении интервала связи у термометра, данный процесс носит немного другой характер. Связь с термометром просто становится нестабильной и хаотично проявляются описанные эффекты.

AndreyGrech, 2023-12-21:
Проблему скорее всего не исправят, у создателя преоритет стоит на проблемы связанные с работой  home assistant и tuya шлюза у создателя нет

pvvx, 2023-12-21:
Это не проблема шлюза Tuya.
Это проблема в коде всех Tuya роутеров. Tuya и Xiaomi не поддерживает Zigbee 3.0 - у них свои стандарты.
А по поводу шлюзов Tuya - Tuya решили закрыть сторонние варианты шлюзов в продаже под своей маркой и будут изготавливать только свои шлюзы, со своими стандартами.

kawadik55, 2025-05-15:
У меня шлюзы/розетки другие, но проблема отвала термометров такая же. Координатор на UPS живет, а роутеры если обесточить на час (термометры к роутерам цепляются), то термометры как будто засыпают после нескольких неудачных коннектов.

pvvx, 2025-05-15:
>  а роутеры если обесточить на час (термометры к роутерам цепляются), то термометры как будто засыпают после нескольких неудачных коннектов.

Не засыпают, а пытаются найти сеть с интервалом в 240 секунд. Если чаще - тогда батарейка кончится через пару часов.
А если засыпают - это им дали команду отключения.
В сети Zigbee есть интервал, через который устройство выкидывается из сети если с ним не было связи. Он устанавливается в роутерах и координаторе. В ZHA почему-то стоит 55 минут.

pvvx, 2025-05-15:
> Координатор на UPS живет, 

А связи у термометра с ним нет, т.к. большая дистанция и ранее связь была через роутер. Вам нужен автономный  роутер, а не координатор с UPS.

kawadik55, 2025-05-16:
Спасибо!
Казалось бы, да, любое ZED устройство должно поискать себе нового Родителя, если что... Но нет...
Провел эксперимент:
контроллер - z2m, термометр заполз на роутер1, рядом расположен еще один роутер2, оба работают от сети.
Выдергиваю роутер1 на 50 мин - на карте z2m роутер1 и термометр болтаются без связей, термометр не искал себе нового Родителя, никуда не привязался. Экран у термометра застыл на последнем показании.

pvvx, 2025-05-16:
Я прошивку из данного репозитория рассматривал второпях и давно. И не помню, происходи ли там обновление экрана при потере связи. В своих, ZigbeeTlc, встроено включение доп. таймера при потере связи, т.к. в SDK основной цикл и прочие процедуры не вызываются, а отрабатывают только события назначенных на софт-таймер.
И в ZigbeeTlc устройства с дополнительными сегментами показывают символом BLE или сопряжения (если есть на дисплее) что сеть потеряна. А переключения на другой работающий роутер всегда происходят...

kawadik55, 2025-05-17:
> Всё это неисправимые болячки Zigbee сетей и без автономного роутера сплошные беды.

Спасибо! Завел автономный, теперь хочу залить прошивку ZegbeeTlc (0x020a) по OTA z2m. Но номера версий совпадают (285224961) с имеющейся db. Через понижение пробовать? Где раздобыть иную версию?

pvvx, 2025-05-17:
285224961 -> 0x11003001
Переходные [прошивки в ZigbeeTLc тут.](https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis)
[db15-0203-79993001-Z03MMC_v0128.zigbee](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/devbis/db15-0203-79993001-Z03MMC_v0128.zigbee)

kawadik55, 2025-05-17:
Благодарю Вас! Поставил. Действительно, теперь термометр весело и задорно прыгает по роутерам при потере своего Родителя! Как и принято в приличных домах...