devbis/z03mmc #96: HW Revision B2.0 - is it supported?

kristoficko, 2023-12-18:
Hi,

I have obtained "a lot" (8pcs) of LYWSD03MMC from Aliexpress for my home automation project.

Tried OTA flashing 3 of them, all 3 got bricket after flashing transitional firmware ATC_ota_40000.bin.

Flashed the latest z03mmc.bin firmware over UART, 1 device got unbricked, 2 still not booting.

This one device is teling me it's unsupported in z2m (ver. 1.34.0 commit: aae7312), same situation after introducing the external_converter with device description into z2m.

`Zigbee2MQTT:info  2023-12-18 15:11:36: 0xa4c138eec99ed547 (0xa4c138eec99ed547): Not supported (Unknown)
Zigbee2MQTT:info  2023-12-18 15:12:31: Starting interview of '0xa4c138eec99ed547'
Zigbee2MQTT:info  2023-12-18 15:12:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eec99ed547","ieee_address":"0xa4c138eec99ed547","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:error 2023-12-18 15:13:31: Failed to interview '0xa4c138eec99ed547', device has not successfully been paired
Zigbee2MQTT:info  2023-12-18 15:13:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138eec99ed547","ieee_address":"0xa4c138eec99ed547","status":"failed"},"type":"device_interview"}'
`

I noticed the HW revision being B2.0 both printed on PCB and mentioned in the OTA flasher log:

`4:18:36 PM: Connecting to: LYWSD03MMC
4:18:51 PM: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 1, CtrID: 0x5830 Not bound, Standard certification, MAC: A4C138FEF129, Capability: 0x28 - front binding, IO: 0x0001
4:20:02 PM: Reconnect
4:20:03 PM: Hardware Revision String: B2.0
4:20:03 PM: Software Revision String: 0130
4:20:03 PM: Firmware Revision String: 1.0.0_0130
4:20:03 PM: Detected Mi device
4:20:03 PM: Connected`

Kind regards

Mike

devbis, 2023-12-21:
Hi! yes, B2.0 is fully supported. According to logs "Failed to interview '0xa4c138eec99ed547'" the device haven't passed the interview correctly. You may need to re-join it until it passes all initial steps.

kristoficko, 2023-12-29:
Hi, I tried re-joining the device many times with no luck. Eventually, re-flashing the same bin file using UART once again helped. This way I managed to un-brick 6 out of 8 devices (OTA flashing the ATC firmware is a 100% brixk success). 

Current status: 
- 6 devices correctly recognized in z2m and reporting (without the use of external converter)
- 2 devices showing values on the display, but can't force them into pairing mode - RST+GND not working, is there any other way how to do this? Re-flashed multiple times, doesn't help.

madison2, 2024-01-02:
> Hi, I tried re-joining the device many times with no luck. Eventually, re-flashing the same bin file using UART once again helped. This way I managed to un-brick 6 out of 8 devices (OTA flashing the ATC firmware is a 100% brick success).
> 
> Current status:
> 
> * 6 devices correctly recognized in z2m and reporting (without the use of external converter)
> * 2 devices showing values on the display, but can't force them into pairing mode - RST+GND not working, is there any other way how to do this? Re-flashed multiple times, doesn't help.

Were you able to connect?
If not, try replacing the battery with a new one and it should work. That's what I did and it worked.