devbis/z03mmc #98: HW Revision B1.5 supported?

morenod, 2023-12-22:
I have flashed all my old sensors with ZigBee firmware and all of them are ok.

I have just receive new ones, and after flashing with the ATC firmware, when trying to flash to ZigBee, I get this error

 Log

15:46:57: Searching for devices
15:47:01: Connecting to: ATC_AD5B91
15:47:09: Hardware Revision String: B1.5
15:47:09: Software Revision String: V4.4
15:47:09: Firmware Revision String: github.com/pvvx
15:47:09: Detected custom Firmware
15:47:11: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.4, Sensor: SHTC3 (SHTV3)
15:47:11: Custom config HEX string: 55051000002804a931318ab4
15:47:23: File: z03mmc.bin
15:47:23: File size: 130004 bytes
15:47:23: Count: 8126
15:47:26: Start DFU
15:47:31: Update error: NotSupportedError: GATT Error Unknown.
15:47:31: Disconnected.
15:47:48: Searching for devices
15:47:52: Connecting to: ATC_AD5B91
15:48:22: NetworkError: Connection Error: Connection attempt failed.
15:48:22: Reconnect 1 from 5
15:48:35: Hardware Revision String: B1.5
15:48:35: Software Revision String: V4.4
15:48:35: Firmware Revision String: github.com/pvvx
15:48:35: Detected custom Firmware
15:48:37: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.4, Sensor: SHTC3 (SHTV3)
15:48:37: Custom config HEX string: 55051000002804a931318ab4
15:50:18: Start DFU
15:50:24: Update error: NotSupportedError: GATT Error Unknown.
15:50:24: Disconnected.

Do you anything to add this HW version to supported ones?

devbis, 2023-12-22:
Firmware update in pvvx's firmware was broken till recent updates about two weeks ago. You may try to update ble firmware to the latest version, if you could.

In case you stuck, you can write firmware over the wires, zigbee or ble, whatever

morenod, 2023-12-23:
which is the latest version to try? Im running the firmware from this file https://github.com/devbis/z03mmc/releases/download/1.0.6/ATC_ota_40000.bin

that fw is running fine, I can get it configured on HASS as xiaomi thermometer, and I see the BT advertise ATC_xxxx when trying to patch the zigbee firmware https://github.com/devbis/z03mmc/releases/download/1.0.6/z03mmc.bin using https://devbis.github.io/telink-zigbee/

devbis, 2023-12-25:
@morenod It is the latest. But I updated it several days ago and added fixes to the original pvvx's firmware.
If you loaded the latest intermediate fw, it will show
Firmware Revision String: github.com/devbis instead of the original strings

morenod, 2023-12-25:
Ok, I will try to download it again, thanks!!

morenod, 2023-12-25:
Cool, reviewed again the document, and flashed with latest files now it is working!! 

Thank you for this incredible work

closed 2023-12-25 by morenod