devbis/z03mmc #99: OTA failed.

Arytmia, 2023-12-22:
Hello,

Firmware date
20231203
Firmware version
3001-0114

I tried to do OTA via Zigbe2mqtt, set local json

But it returns an error
Failed to check if update available for '0xa4c138c2457abc2b' (No image available for imageType '522')

When it adds in the json imagetype, it gets the error
Failed to check if update available for '0xa4c138c2457abc2b' (Unexpected string in JSON at position 93)

This error indicates errors on different lines depending on the file used.

I would like to switch back to the bluetooth version.
Please help me find the correct file.

devbis, 2023-12-22:
You have probably flashed firmware from repository https://github.com/pvvx/ZigbeeTLc
I s a fork, thought it has a different versioning.
I suppose, you should use 1141-020a-99993001-ATC_v46.zigbee  (020a!)  from this folder https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

Arytmia, 2023-12-22:
Still errors in lines of code.

I can't figure out what version I installed 🙄.

I am left with only usb->uart?

devbis, 2023-12-26:
It looks like you misconfigured your local ota, maybe ota  json is badly formatted. You can always use UART to flash any firmware.