devbis/z03mmc #102: battery life very short

carlituxman, 2024-01-05:
The battery life is very short. Is possible change the read sensor period?

devbis, 2024-01-05:
What firmware version do you use?

carlituxman, 2024-01-05:
Fw version 3001-0118

How can I revert to bluetooth?

devbis, 2024-01-06:
It's not a firmware from this project, but from a fork. Please, use right firmware from https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota and local OTA functionality for z2m or ZHA. Or solder wires if you're using different software.

closed 2024-01-06 by devbis

carlituxman, 2024-01-06:
I just revert to ble version and then install: fw:3001-1006 but at z2m still appear 3001-0118
I sure installed https://github.com/devbis/z03mmc/releases/tag/1.0.6

devbis, 2024-01-06:
Z2m caches some data from devices, you can read values from attributes manually via developer console for the device. Or reconfigure (don't know if it helps), or delete the device, restart z2m and pair again. In case of a different tool to control zigbee network there could be different ways to clean the cache

pbranly, 2024-01-07:
Hi @devbis 
Sorry for my English.
I am a little bit lost with all the versions . I use your version (1.0.6) for my 2 thermometers 
Is your advice to use a firmware from a different GitHub ?
Thanks
Phil

devbis, 2024-01-07:
No.

pbranly, 2024-01-07:
> No.

👍