devbis/z03mmc #103: LYWSD03MMC keeps reporting N/A for temperature and humidity

MathijsG, 2024-01-07:
I recently flashed my LYWSD03MMC with z03mmc.bin to transfer it from the custom BLE firmware to Zigbee.

It pairs fine and everything, reports LQI, visually switches to Fahrenheit/Celsius, but the reported temperature and humidity keeps reporting N/A.

I've tried re-pairing them, removing them. Using a different coordinator didn't help as well.

My last resort is reflashing it back to Bluetooth via UART.

Any direction to pinpoint this issue?

The device physically looks like this:
[image]

devbis, 2024-01-07:
What zigbee coordinator software do you use? What version of firmware did you write?

pvvx, 2024-01-08:
What does "RECONFIGURE" show?
[image]

MathijsG, 2024-01-08:
Sorry I'm in the Zigbee2mqtt camp. Should make that more clear next time, sorry.

I did it in Z2M however:

[image]

I took this z03mmc.bin file from: https://github.com/devbis/z03mmc/releases/tag/1.0.6

EinSchwerd, 2024-01-11:
I'm facing a similar issue. 
I force-removed the device from Z2M a couple of times and it recovered I was getting proper numbers. 
However, after some time the thermometer locks up. It stops sending any values and it stops updating the temperature on the display.
[image]

I OTA flashed it mid-December 2023 with the latest available firmware and it was working OK for a while.
I'm using a SONOFF Zigbee 3.0 USB Dongle Plus with Debian 12 and Home Assistant 2024.1

MathijsG, 2024-01-11:
@EinSchwerd if you get actual numbers for temperature and humidity at the zigbee-level then this seems like a different issue. Maybe yours has something to do with distance towards the device? Move it closer to the actual device it is connected to, or pair it to something else closer?

EinSchwerd, 2024-01-12:
I don't get numbers on zigbee-level. I don't get numbers at all. It seems like the device is deadlocked as the server does not receive any messages other than that the thermometer is connected.

devbis, 2024-01-12:
Can you read data from attribute manually? Temperature/measuredValue

EinSchwerd, 2024-01-14:
I put it onto the coordinator antenna and no change. How can I manually read the value from it? It does not change the value displayed on the device itself regularly. It only does a value update when I re-insert the battery.

ildus, 2024-01-14:
Facing same issue, using the firmware from 09.11.2023. The values showed as 'N/A' in zigbee2mqtt.

Yari117, 2024-01-23:
Same here! Works for some time and then stops to respond. Taking the battery out for 10 sec and placing it back helps - new values are transmitted for some period (several days) and then the same thing happens. Z2M with external converter.

MathijsG, 2024-02-10:
When I compare the amount of temp updates to my still-Bluetooth temp sensors (exact same devices):
Zigbee variant:
[image]

Bluetooth variant (custom firmware):
[image]

It seems like the Bluetooth version updates much more frequently. Is this something I can change on my side? The paramaters of reporting an update are already quite low (often) by default, yet it seems like it's not reporting temp changes as accurately and timely as my Bluetooth sensors.

devbis, 2024-02-10:
You can configure the reporting time and delta using standard zigbee functionality. What software do you use?

MathijsG, 2024-02-10:
I use Zigbee2mqtt. And the defaults for this device are already very low in reporting (so it's often reporting al ready). See attached screenshot:
[image]

But it seems more like the sensor ignores these settings somehow, as it's often keeping the same value. Sometimes it only updates when I pressed these buttons (the refresh icons):
[image]

devbis, 2024-02-10:
The configuration is correct. Try upgrade to the latest firmware, use fresh battery and re-join if needed.

pvvx, 2024-02-11:
Один и тот-же таймер `app_reportMinAttrTimerCb` переназначается на все события:
https://github.com/devbis/z03mmc/blob/master/src/reporting.c#L80C87-L80C111
В итоге отрабатывает одно.

devbis, 2024-02-11:
Это не таймер, а колбек на таймер. О каких ограничениях в коде идет речь, если это вызывает такой эффект? Есть ссылка на код реализации таймеров, где это вызывает проблемы?

pvvx, 2024-02-11:
А как назначается таймер?
Именно по адресу callback

pvvx, 2024-02-11:
Я не разбирал ваш код, но эффект выходит именно такой – когда значения T и H меняются быстро, передается только одно T или H.

Yari117, 2024-03-21:
> Recently there was an OTA update for my sensor, and that fixed it :-)!

OK, I thought the OTA helped, but the sensor keeps dropping out. Battery out reset helps for a day or so, then the sensor stops reporting

DiegoJp, 2024-06-07:
Happened to me twice between last week and today. New CR2032 battery and all. Device apparently crashes and keeps stuck on a fixed value of T° and hum%. The workaround as mentioned above is just taking out the battery and put it again. But that's no good for the plastic of the cover which is a bit hard to take out.