devbis/z03mmc #107: OTA Images currently unavailable

memphisraynz, 2024-01-15:
Hi,
I migrated these devices across to Zigbee, and all working well except for being able to update a device on an older firmware.

It is currently listed as firmware - 3001-0117
 when checking for updates I get this error: 
OTA: Images currently unavailable for device 'LYWSD03MMC', hardwareVersion '14', manufacturerName Xiaomi, {"fieldControl":0,"manufacturerCode":4417,"imageType":522,"fileVersion":18362369}

Any idea on what I can do?
Thanks

devbis, 2024-01-15:
Hi! This firmware is NOT from this project, the latest version is 3001-1006 and imageType 515.

memphisraynz, 2024-01-15:
Appreciate it, I will flash the BLE version then back to this repository.

closed 2024-02-04 by devbis