devbis/z03mmc #108: Can not flash OTA the transitional firmware ATC_ota_400000

christescu, 2024-01-20:
When selecting ATC_ota_400000, I get the message 'Size firmware is more 208 kbytes!'
I have custom firmware:
Hardware Revision String: B1.9
Software Revision String: V4.1
Firmware Revision String: github.com/pvvx
Detected custom Firmware
Hardware Version: LYWSD03MMC B1.9, Software Version: 4.1, Sensor: SHT4x
Custom config HEX string: 552d01fdf42804a93131033c000000

Same thing happens if I first flash 'Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin', then I make 'Do Activation'.

devbis, 2024-01-21:
Transitional firmware is just about 80kb. Maybe you have downloaded something different.

christescu, 2024-01-21:
Right, I take the file with a RPI and yes, it has 81k.
I don't know why windows download it as a 387k file.

closed 2024-01-21 by christescu