devbis/z03mmc #112: new z2m features refused

pbranly, 2024-02-02:
Hi
How is it possible to use the new features that seem available in zigbee2mqtt ? each one is refused as unsupported
Thanks
Philippe
[image]

polygon242, 2024-02-03:
same here
`Publish 'set' 'temperature_calibration' to 'Sensor Temp Xiaomi 05' failed: 'Error: Write 0xa4c1384a104079e6/1 hvacUserInterfaceCfg({"256":{"value":1,"type":40}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

devbis, 2024-02-03:
The convertor is broken by a user. I"m working on it. Probably, a custom convertor from this repo could help.

pbranly, 2024-02-03:
Hi
Will you make a new release ?
Thanks
Phil

VivantSenior, 2024-02-04:
> The converter is broken by a user. I"m working on it. Probably, a custom converter from this repo could help.

@devbis thanks for the hint. I was able to use [external converter](https://github.com/devbis/z03mmc/blob/master/converters/lywsd03mmc.js) from your repo. Can't see any errors coming from my instances of [LYWSD03MMC](https://www.zigbee2mqtt.io/devices/LYWSD03MMC.html) and all of the sensors are correctly reporting data.

pbranly, 2024-02-04:
Hi after several tentatives I could use the external converter 
The result is that all the additional menus in my initial post are now absent 
Is it normal ?
Thanks
Phil

devbis, 2024-02-04:
The fixed convertor is in [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7015) but may be delayed due to z2m release policy.
It adds new features which were introduced in the latest FW 1.1.0 with backward compatibility. 

Before it it merged to z2m and released, you can use custom convertor from this repo: https://github.com/devbis/z03mmc/blob/master/converters/lywsd03mmc.js

pbranly, 2024-02-04:
Hi 
I don't have any proposal to update the f/w to 1.1.0.
So with the new converter, the new configuration menus are back but generate a lot of errors ...so don't work
How to force f/w update to 1.1.0 ?
Thanks
Phil

devbis, 2024-02-05:
New features will be available starting from 1.1.0. To have it in z2m you can either wait for https://github.com/Koenkk/zigbee-OTA/pull/440 merged or use local OTA index https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

closed 2024-02-06 by pbranly