devbis/z03mmc #113: How does the ota update work?

carlituxman, 2024-02-05:
Just release 1.1.0 and I have 1.0.6 (3001-1006) but not detect OTA update to 1.1.0.

How works detect ota update?

Thanks

devbis, 2024-02-05:
Please, track https://github.com/Koenkk/zigbee-OTA/pull/440

dhewg, 2024-02-06:
Sorry for reusing this issue, but is it possible to ota from a pvvx fw? ZHA setup here

devbis, 2024-02-06:
@dhewg You may use this file (unzip it) to flash version 1.1.0 over pvvx's firmware with a standard ZHA OTA. Haven't tried it but maybe this article will help you https://sonoff.tech/product-review/tutorial/local-configure-zigbee-ota-in-home-assistant-tutorial/

[1141-020a-11003001-z03mmc.zigbee.zip](https://github.com/devbis/z03mmc/files/14179644/1141-020a-11003001-z03mmc.zigbee.zip)

dhewg, 2024-02-06:
Nice, thanks!
I've ota'ed before, so generally it's working. I've asked because updating only works when the manufacturer code matches? But it seems you have that covered ;)
At least it triggers an ota with a running pvvx fw and that file `[zigpy.zcl] [0x0DE8:1:0x0019] Updating: Xiaomi LYWSD03MMC`
Let's see if that succeeds...

dhewg, 2024-02-06:
For the record, HA 2024.1 fails for me with a timeout, but a 2024.2 beta (which reworked zha ota) works fine:
[image]

dhewg, 2024-02-06:
I can't set some config attributes with zha:
[image]
Is that a zha thing or are the missing attributes not announced or something like that?

shirou93, 2024-02-06:
@devbis after flash 020a frimware from You next updates will work on standard z2m ota?

devbis, 2024-02-07:
@lbarjak Device is sleeping. Try one again, replug the battery.

lbarjak, 2024-02-07:
Yes, works now, thx.
The new firmware shows the old 3001-1006 number in zigbee2mqtt.

Ivan Belokobylskiy ***@***.***> ezt írta (időpont: 2024.
febr. 7., Sze, 13:28):

> @lbarjak <https://github.com/lbarjak> Device is sleeping. Try one again,
> replug the battery.
>
> <https://github.com/devbis/z03mmc/issues/113#issuecomment-1931944890>, or
> unsubscribe

carlituxman, 2024-02-07:
Try in dev console in zigbee2mqtt: cluster -> genBasic->swBuildId and read. Maybe it's cached

gdbd, 2024-02-09:
updated via ota one of them today, version not changes and still  	20231109	3001-1006

then updated zigbee2mqtt on 1.35.3 

removed custom external converter

nothing changes...

what i am doing wrong?

thermosiphonas, 2024-03-29:
For the love of me, I cannot figure out how to update these devices!

I put 

> zha:
      zigpy_config:
           ota:
      otau_directory: /config/zigbee_ota

into my confifuration.yaml

download 1141-0203-11003001-z03mmc.zigbee and z03mmc.bin and put them in the zigbee_ota directory

Then in ZHA I go to Manage Zigbee Device > Clusters > Ota > Commands > Image Notify and get the error

> [140151761094464]` '>=' not supported between instances of 'NoneType' and 'PayloadType'

I have also tried through ZHA Toolkit but still nothing happens

devbis, 2024-04-05:
@lllluke  You can use this file to convert from pvvx's zigbee firmware https://github.com/devbis/z03mmc/issues/118#issuecomment-1942371299

lllluke, 2024-04-05:
Thank you, I already tried that but the flasher doesn't find the device anymore. So I thought OTA via z2m was the only way to go.

jksmurf, 2024-10-14:
I am struggling with this revert from Zigbee to BLE.

I have tried 

- 1141-020a-11003001-z03mmc.zigbee
- 1141-020a-99993001-ATC_v47.zigbee
- db15-0203-99993001-ATC_v47.zigbee
-

I keep getting the message that the FW is newer than the latest one online? 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]