devbis/z03mmc #114: Manufacturer ID has changed after upgrade to 3001-1100

maaad, 2024-02-05:
After upgrading to 3001-1100(20240204) via custom OTA 'Manufacturer ID' and 'Image Type' has changed to:
`Read result of 'genOta': {"manufacturerId":56085,"imageTypeId":65535}`
Is that expected result?

pbranly, 2024-02-05:
Hi 
Could you explain how you did this upgrade ? 
Thanks
Phil

devbis, 2024-02-05:
@maaad  Yes, it is the expected result. I will use Manufacturer ID 0xdb15 from now on to reduce errors because of confusions when using firmware from other forks.

pbranly, 2024-02-05:
Thank you but which file did you use ? 
I tried one from z2m but it does not generate new OTA for the thermometer 
Thanks
Philippe

maaad, 2024-02-05:
this one: https://github.com/devbis/z03mmc/releases/download/1.1.0/1141-0203-11003001-z03mmc.zigbee

devbis, 2024-02-05:
For some reason, image type is 0xff. Investigating

pbranly, 2024-02-05:
> this one: https://github.com/devbis/z03mmc/releases/download/1.1.0/1141-0203-11003001-z03mmc.zigbee

Thanks 
I could upgrade 2 devices 
One device dead at the end .... no more screen
I will try direct upload 
Phil

devbis, 2024-02-05:
@pbranly please wait until i fix the image type issue

Soukyuu, 2024-02-05:
Will updating to the new manufacturer id also fix the issue on the current release of z2m where some functionality for an alternative got merged and now spamming my log with errors?

pbranly, 2024-02-05:
> @pbranly please wait until i fix the image type issue

Hi too late ! 2 of my 3 devices work fine with the latest f/W.
As soon as your new one will be available, I will re-OTA 
I will also try to reanimate the third one with direct upload 
Thanks for your work
Phil

devbis, 2024-02-05:
My investigation showed that it wasn't an issue. New SDK from Telink works this way. Default value is 0xffff but when update is in progress it shows correct values. Will leave it as is.

closed 2024-02-05 by devbis

PabloPL, 2024-02-20:
I have the same issue. After latest update i'm getting
```
ead result of 'genOta': {"imageTypeId":65535,"manufacturerId":56085}
```
And when trying to go back to BT version i can see
```
Debug 2024-02-20 21:52:18 Received Zigbee message from 'Xiaomi sensor', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285224961,"imageType":515,"manufacturerCode":56085}' from endpoint 1 with groupID 0
Debug 2024-02-20 21:52:18 Got OTA request '{"fieldControl":0,"manufacturerCode":56085,"imageType":515,"fileVersion":285224961}'
Debug 2024-02-20 21:52:18 ZigbeeOTA: downloaded main index
Debug 2024-02-20 21:52:18 ZigbeeOTA: Loading override index /etc/zigbee2mqtt/ota_xiaomi.json
Debug 2024-02-20 21:52:18 ZigbeeOTA: getting local firmware file /etc/zigbee2mqtt/ota/1141-0203-99993001-ATC_v46.zigbee
Debug 2024-02-20 21:52:18 getNewImage for '0xa4c13849c85d6bd0', meta {"fileVersion":2576953345,"url":"/etc/zigbee2mqtt/ota/1141-0203-99993001-ATC_v46.zigbee","force":true}
Debug 2024-02-20 21:52:18 ZigbeeOTA: getting local firmware file /etc/zigbee2mqtt/ota/1141-0203-99993001-ATC_v46.zigbee
Debug 2024-02-20 21:52:18 getNewImage for '0xa4c13849c85d6bd0', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4417,"imageType":515,"fileVersion":2576953345,"zigbeeStackVersion":2,"otaHeaderString":"Telink OTA BLE device\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":82674}
Debug 2024-02-20 21:52:18 Update of 'Xiaomi sensor' failed (AssertionError [ERR_ASSERTION]: Manufacturer code mismatch)
```
@devbis Do You have any idea how to fix this issue?
The best would be without flashing by uart - i don't have it yet setup, also i have a few devices to handle.

PabloPL, 2024-02-21:
@devbis  Thank You!
OTA update went without issues and i was able to pair sensor using OpenMQTT over bluetooth.
Thanks again.

devbis, 2024-02-21:
For local ota only url and force are required.