devbis/z03mmc #116: LYWSD03MMC B1.4 bricked

ddzobov, 2024-02-06:
Just bricked 2pcs of LYWSD03MMC
Hardware Version: LYWSD03MMC B1.4, Software Version: 9.9, Sensor: SHTC3 (SHTV3)

I'm doing flashing as described in readme:
1) ATC_ota_400000
2) reconnect with new name
3) z03mmc.bin

But after flashing target firmware device stucks on values and not connecting to zigbee.
Reinserting battery causes showing initial values, then screen blinks and stucks again (not updating values and not connecting to zigbee)

realdigger, 2024-02-06:
Try to replace batteries with new ones.

Sch8er, 2024-02-07:
I did not got a new name after flashing with `ATC_ota_40000.bin`  and replacing the battery afterwards (for minutes). I used the parameters from `token_extractor` to be able to `Login`. `Do Activation` did not work -  wait for 10 minutes and nothing happend (maybe timeout)

13:12:22: Searching for devices
13:13:25: Connecting to: LYWSD03MMC
13:13:32: Hardware Revision String: B1.4
13:13:32: Software Revision String: 0159
13:13:32: Firmware Revision String: 2.1.1_0159
13:13:32: Detected Mi device
13:13:32: Connected
13:14:00: Send Login, please wait...
13:14:01: Received device infos are correct
13:14:01: Login successfull
13:14:08: File: ATC_ota_40000.bin
13:14:08: File size: 82516 bytes
13:14:08: Count: 5158
13:14:10: Start DFU
13:15:56: Update done after 104.879 seconds
13:16:03: Disconnected.

So flashing with `z03mmc.bin` failed at the end

14:40:56: Searching for devices
14:41:11: Connecting to: LYWSD03MMC
14:41:18: Hardware Revision String: B1.4
14:41:18: Software Revision String: 0159
14:41:18: Firmware Revision String: 2.1.1_0159
14:41:18: Detected Mi device
14:41:18: Connected
14:41:32: Send Login, please wait...
14:41:32: Received device infos are correct
14:41:32: Login successfull
14:41:39: Start DFU
-> stopped here [image]
14:44:01: Update error: NotSupportedError: GATT operation failed for unknown reason.
14:44:01: Disconnected.

Tried at different device (mobile phone), too. 

Any ideas?

ddzobov, 2024-02-07:
> Try to replace batteries with new ones.

Just replaced battery and it works. Very strange thing, maybe add this to docs?

realdigger, 2024-02-07:
> > Try to replace batteries with new ones.
> 
> Just replaced battery and it works. Very strange thing, maybe add this to docs?

Perhaps zigbee firmware is more sensitive to battery level.

ddzobov, 2024-02-07:
> > > Try to replace batteries with new ones.
> > 
> > 
> > Just replaced battery and it works. Very strange thing, maybe add this to docs?
> 
> Perhaps zigbee firmware is more sensitive to battery level.

I flashed third sensor with https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html with same battery (after flashing previous two) and it connected after flashing without problems.

First bricked sensor was thrown away (i tried to flash it via UART and completely bricked it), second was connected after battery replacement with 100% capacity. 

What is the difference between these two firmwares?

mikekarpov, 2024-02-09:
After flashing ATC_ota_40000.bin now bricked. I'm trying to program via LSR825xComFlasher.py it but I get an error 

C:\Users\min\Downloads>python3 TLSR825xComFlasher.py -p com6 -t5000 wf 0 ATC_ota_40000.bin                                          =======================================================                                                                             TLSR825x Flasher version 00.00.04                                                                                                   -------------------------------------------------------                                                                             Open com6, 921600 baud...                                                                                                           Reset module (RTS low)...                                                                                                           Activate (5000 ms)...                                                                                                               Chip sleep? -> Use reset chip (RTS-RST): see option --tact

13werwolf13, 2024-09-30:
I have a similar problem

I bought two identical `LYWSD03MMC`
one flashed without problems to the new firmware
the second one logged `NetworkError: GATT Server is disconnected` and shut down. I tried changing the battery, but it doesn't help, there's nothing on the screen and ZHA doesn't see it.