devbis/z03mmc #118: LYWSD03MMC-z

Vortecsz, 2024-02-12:
After flashing the LYWSD03MMC  with z03mmc 1.1.0 the device connect's to zigbee2mqtt.
Flashed first from OTA and after from UART but the problem persist.
The only available value is the battery, it is strange because the device is called: LYWSD03MMC-z
Other's users have also this problem: https://www.reddit.com/r/homeassistant/comments/1aok4l4/help_im_stuck_with_an_unknown_zigbee_device/

devbis, 2024-02-13:
Hi! You haven't flashed firmware from this repo but from a fork https://github.com/pvvx/ZigbeeTLc/

You can follow up with that fw, or use local ota to convert to my firmware using local OTA (z2m or ZHA) and  [1141-020a-11003001-z03mmc.zigbee.zip](https://github.com/devbis/z03mmc/files/14179644/1141-020a-11003001-z03mmc.zigbee.zip) as new firmware.

toelke, 2024-02-16:
I had the same issue. This is what I did:

1. I copied the file `custom_devices_diy.js` from the running docker container: `docker cp zigbee2mqtt:/app/node_modules/zigbee-herdsman-converters/devices/custom_devices_diy.js .`
2. I modified that file by replacing `LYWSD03MMC` with `LYWSD03MMC-z` for `zigbeeModel`.
3. I mounted that modified file into the same place in the docker-container `docker run ... -v ./custom_devices_diy.js:/app/node_modules/zigbee-herdsman-converters/devices/custom_devices_diy.js ...`
4. Now zigbee2mqtt would correctly detect the device, but obviously that is a bit brittle.

When I do an OTA now, the device is updated to a newer version of the fork :-(

I tried adding the zip-file you linked to a local index, but z2m does not pick up that file but still the fork. Do you know how I can force z2m to OTA your file?

configuration.yaml:
```
ota:
    zigbee_ota_override_index_location: my_index.json
```

my_index.json

```
[{
        "url": "1141-020a-11003001-z03mmc.zigbee",
        "force": "true"
}]
```

devbis, 2024-02-16:
Your ota configuration looks good. It depends on errors that you have trying to update. 
z2m asks device for possible manufacturer and image type and then checks its index. 
Did you unzip the file? Is it located in the same folder as configuration.yaml and my_index.json?

toelke, 2024-02-16:
It is, yes.

drag0n141, 2024-02-16:
Hey,
i have the same problem, after flashing the [1141-020a-11003001-z03mmc.zigbee.zip] (https://github.com/devbis/z03mmc/files/14179644/1141-020a-11003001-z03mmc.zigbee.zip) file it has now the same version as my others. But is shows still the "LYWSD03MMC-z" name and not all functions. Is it possible to reflash it without the -z?

devbis, 2024-02-16:
z2m caches device properties if mac address wasn't changed. You can reread it via dev console, or remove device, restart z2m and join the device again.

toelke, 2024-02-16:
Indeed. Dev console -> cluster genBasic -> attribute modelId -> read.

Thanks!

Milinator, 2024-02-23:
> Hi! You haven't flashed firmware from this repo but from a fork https://github.com/pvvx/ZigbeeTLc/
>
> You can follow up with that fw, or use local ota to convert to my firmware using local OTA (z2m or [...]

I was having the exact same issue and I was trying to flash different OTA updates (but they never showed up, because they were probably/obviously not the right ones). Just out of curiosity: How did you know the right update firmware file?

MathijsG, 2024-04-03:
In the end I went back to Bluetooth via: https://github.com/devbis/z03mmc?tab=readme-ov-file#return-to-bluetooth-firmware

Using the file: https://github.com/devbis/z03mmc/blob/master/assets/db15-0203-99993001-ATC_v46.zigbee

Why did I went back? Well, the temperature almost never seemed to update. It kept the current temperature until I clicked on the refresh icons of the temperature in Zigbee2mqtt.

darozmini, 2025-06-15:
I flashed LYWSD03MMC with 1141-020a-01283001-Z03MMC.zigbee and can't connect to aqara hub m3 nor bluetooth. I would be really grateful if anybody could give me a clue what went wrong
20:41:47: Get Advertising MAC (Web Experimental Features!)
20:41:47: Connecting to: LYWSD03MMC
20:41:49: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 19, CtrID: 0x5858 Not bound, Standard certification, MAC: A4C13841DEF7, Data encrypted
20:41:58: Hardware Revision String: B1.4
20:41:58: Software Revision String: 0159
20:41:58: Firmware Revision String: 2.1.1_0159
20:41:58: Detected Mi device
20:41:58: Connected
20:43:48: Send Login, please wait...
20:43:48: Received device infos are correct
20:43:49: Login successful
20:45:28: Load firmware file 'bin/ATC_v52.bin'...
20:45:28: File: bin/ATC_v52.bin
20:45:28: File size: 86324 bytes
20:45:28: Add signature file.
20:45:28: Count: 5718
20:45:31: Start DFU
20:46:15: Update done after 43.453 seconds
20:46:21: Disconnected.
20:46:30: Reconnect to: LYWSD03MMC
20:46:32: Hardware Revision String: B1.4
20:46:32: Software Revision String: V5.2
20:46:32: Firmware Revision String: github.com/pvvx
20:46:32: Detected custom Firmware
20:46:32: Hardware Version: LYWSD03MMC B1.4, Software Version: 5.2
20:46:32: Custom config HEX string: 5507100a062804a9313100b4
20:53:49: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee'...
20:53:49: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee
20:53:49: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01283001
20:53:49: APP Release: 0.1 Build: 2.8, Stack Version: 3.0 Release: 0.1
20:53:49: File size: 131364 bytes
20:53:49: Count: 8211
20:53:53: Start DFU
20:53:55: Clear Ext.OTA region (129kib)...
20:53:57: Waiting Start ext.OTA...
20:54:03: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
20:54:47: Update done after 43.362 seconds
20:54:51: Disconnected.
20:54:57: Reconnect to: LYWSD03MMC
20:55:05: NetworkError: Connection Error: Connection attempt failed.