devbis/z03mmc #126: battery voltage?

mr-aleks, 2024-02-27:
Tell me how to display battery voltage? In the stock firmware it was...
I recently started using z2m and maybe I don’t know something, please tell me.
My settings:
[image]
[image]

devbis, 2024-02-28:
It was excluded from z2m converted as redundant property. But you may use custom converter with added voltage

mr-aleks, 2024-02-28:
I added the converter from this article but there is still no voltage...
https://smarthomescene.com/guides/convert-xiaomi-lywsd03mmc-from-bluetooth-to-zigbee/
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['LYWSD03MMC'],
    model: 'LYWSD03MMC',
    vendor: 'Xiaomi',
    description: 'Temperature & humidity sensor',
    fromZigbee: [fz.temperature, fz.humidity, fz.battery],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.temperature(endpoint, {min: 10, max: 300, change: 10});
        await reporting.humidity(endpoint, {min: 10, max: 300, change: 50});
        await reporting.batteryVoltage(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.temperature(), e.humidity(), e.battery()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Hunter32R, 2024-02-28:
Add to exposes: e.battery_voltage()

mr-aleks, 2024-02-28:
> Add to exposes: e.battery_voltage()

thank you, it worked

closed 2024-02-28 by mr-aleks