devbis/z03mmc #129: Cannot re-pair device

felipebraz, 2024-03-05:
I flashed my device and in the first time worked just fine im Zigbee2MQTT but I had to change my dongle and now I cannot pair it again.
Other devices (lights, outlets,...) was easy to pair again with the new network, but all Mi Thermometers I've flashed doesn't enter in pairing mode.
I've tried take off the battery (for more than 1 hour), short reset and GND (3, 5, 10 and 15 seconds), pray, light a candle,...
All (3) flashed thermometers show the same behavior.
Any ideas?

devbis, 2024-03-13:
If the version is higher than 1.0.5, repairing in firmware was fixed. You may need to upgrade coordinator firmware. Otherwise, you could try to upgrade firmware to the latest version from this repo.

marclefevre, 2024-03-29:
I'm in the same spot  The device I have to repair shows:
Firmware build date
20231215
Firmware version
3001-0117

Did you ever get a resolution on your issue?
How do you do an OTA update to the current firmware?

felipebraz, 2024-04-01:
I still have the issues. To be fair I didn't have the time to explore solutions yet...

devbis, 2024-04-01:
> I'm in the same spot The device I have to repair shows: Firmware build date 20231215 Firmware version 3001-0117

This firmware is from a fork, not from this repository.

TheNitek, 2024-06-05:
Re-Pairing also didn't work for around half by devices. Using a different (stronger?) battery solved it for my.

marclefevre, 2024-07-07:
> @felipebraz <https://github.com/felipebraz> did you ever solve this?
>
> I have a similar issue, I'm having to recreate my ZigBee network and
> re-pair all my devices.
>
> Mine simply won't go in to pair mode. I've replaced the battery tried
> shorting the GND and RST pins for 3, 5 and 10 seconds. I've also had the
> pins shorted whilst inserting the battery. The device simply won't go into
> pairing mode.
>
> @devbis <https://github.com/devbis> you mentioned updating the device
> firmware. Is there an OTA method of re-installing/updating the firmware if
> I can't get it to pair with my coordinator?

No, I never did find a solution for this.  Furthermore, most of the 10
thermometers I purchased and converted have switched from Fahrenheit back
to Celcius and refuse to go back to F so they have a lot less utility now.

devbis, 2024-07-07:
>  you mentioned updating the device firmware. Is there an OTA method of re-installing/updating the firmware if I can't get it to pair with my coordinator?

No, the only way to flash/update it without connecting to the network is using soldering to sws pin

kitor, 2024-07-11:
> Furthermore, most of the 10 thermometers I purchased and converted have switched from Fahrenheit back to Celcius and refuse to go back to F so they have a lot less utility now.

As I have the problem with pairing right now - it seems if you hold reset to gnd for a few seconds my device switches between celcius and Fahrenheit. 
In my case I was not able to pair device via router, but it paired via coordinator without any issues. ZHA shows my firmware as 0x11003001

devbis, 2024-07-11:
To reset the device you need to short contacts for more than 5 seconds. It is described in the documentation.

dJani97, 2024-08-25:
Same issue here. I tried connecting the GND and Reset pins for >30 seconds, I also tried removing the battery before/during/after connecting the pins, but no luck. I also unplugged my routers, so only the coordinator remains on the network, still no luck.

I'm on `v1.1.0 (ATC_ota_40000.bin)`

dJani97, 2024-08-26:
It turned out that I had been resetting the thermometer incorrectly. The reset works by connecting the two pins (ground & reset), but right next to the pins is the coin cell battery. The metal thingy I used to connect the pins also touched the battery, and touching the pins and the battery at the same time probably short-circuited it 😄 The screen flashed, so I thought the reset was triggered, but no, it wasn't.

After figuring that out, I used an insulated wire to touch ONLY the two pins and nothing else for about 10 seconds, and the device finally entered pairing mode.

alinelena, 2025-01-07:
just to say, i was suffering the same predicaments, resetting these is a pain. in the end the most reliable way was to solder two wires on reset and gnd and do it like that.

alinelena, 2025-01-23:
or https://github.com/devbis/z03mmc/issues/170#issuecomment-2610934980

chrishome7, 2025-02-03:
They don't reset 95% of the time and never seem to go into pairing mode or pair, at least not with z2m.

chrishome7, 2025-02-03:
They don't reboot 95% of  the time. I cant get any to go into pairing mode or pair to z2m. I see this in comment forums everywhere along with battery issues. I wish I would have known about pairing issues prior to flashing.