devbis/z03mmc #133: Expand the project to cover th RUUVI tags which are used quite extensibly?

thestraycat, 2024-03-14:
The ruuvi project has been a great bluetooth beacon project for many years... And currently a hardware device is needed to manage these extend their range for optimal support.

Would it be possible to convert these to use the zigbee protocol at all?

https://ruuvi.com/ruuvitag/

devbis, 2024-03-15:
According to the images, the chip N52832 that is used in ruuvitag doesn't support Zigbee, only BLE
https://www.nordicsemi.com/products/nrf52832

thestraycat, 2024-03-15:
Thanks for confirming, a bit of a shame... Is the custom firmware not applying an emulated zigbee using the underlying hardware then i take it? So in escense, the only reason the xiaomi temp sensors work is that in the spec sheet they say they also support zigbee??

devbis, 2024-03-21:
It says that its sibling CPU supports zigbee. But under the hood, all chips are the same. Maybe nordic chips has something similar and lower models support featured from the models above, but it needs to be checked. I think it is not the matter of this project which aims Telink chips.

closed 2024-03-21 by devbis