devbis/z03mmc #137: high battery consumption

shurakenas, 2024-03-29:
The battery is running out in front of my eyes. Is this normal?
[image]
[image]

devbis, 2024-03-30:
Sometimes, it is required to replug the battery to get the sensor back to a usual sleeping rate

shurakenas, 2024-03-30:
I reconnected the battery, it's still a fast discharge. Do I need to replace the battery with a new one?

shurakenas, 2024-03-30:
When I returned the bluetooth firmware, the battery is 100%.
[image]

OscarAlmgren, 2024-04-01:
Yeah, I have this weird behaviour too. What can be done? 
[image]

propi62, 2024-04-06:
my ultracheap aliexpress battery is still strong after 5 months. however i dont understand the batteryspike on the last days.
[image]

trablin45, 2024-08-12:
> Im having the same problem with a new LYWSD03MMC with brand new batteries, just TWO days from flashing battery went from 100% to 80%. Is this a bug from the battery report or is it really eating up the battery? Any way to fix this?
update on my case, first battery drowned in 1 month 26 days. Replaced and second battery stopped reporting at 1 month 12 days. FW 3001-1100. Duracell batteries. Using Z2mqtt. At these drowning speeds FW is unusable

devbis, 2024-08-12:
@trablin45  You can try a fork https://github.com/pvvx/ZigbeeTLc or revert to BLE using z2m local OTA

trablin45, 2024-08-13:
> @trablin45 You can try a fork https://github.com/pvvx/ZigbeeTLc or revert to BLE using z2m local OTA It is not the normal consumption for the device, connected to coordinator
is it possible to flash the fork from z2m local ota?

devbis, 2024-08-13:
Yes, you can use these binaries: https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis

Na0mir, 2024-08-29:
Hello! Same issue here, I just flashed to zigbee and consuming ~10% per day 😥
[image]

I will try this fork https://github.com/pvvx/ZigbeeTLc
@trablin45  did you tried it ?

Na0mir, 2024-09-30:
> > I will try this fork https://github.com/pvvx/ZigbeeTLc @trablin45 did you tried it ?
> > Yes I did flashed the fork but its too soon to comment on battery, I will keep testing and update

Hi! Any news on it after 3 weeks of usage?

trablin45, 2024-10-01:
> > > I will try this fork https://github.com/pvvx/ZigbeeTLc @trablin45 did you tried it ?
> > > Yes I did flashed the fork but its too soon to comment on battery, I will keep testing and update
> 
> Hi! Any news on it after 3 weeks of usage?

One month eleven days after battery is reporting 97% and operating normally, I can recommend the fork for those with the battery drain problem

nobwyn, 2024-10-11:
@trablin45: which firmware version did you try?

ovidiu-vas, 2025-01-02:
@trablin45 , there's multiple .zigbee files at https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis . Which one did you use for local OTA flash?

shaarkys, 2025-05-08:
Just in case someone would be searching for solution OTA on Zigbee2MQTT... this JSON below worked to revert to ATC BLE for LYWSD03MMC (v [3001-100](https://github.com/devbis/z03mmc/releases/tag/1.1.0))), then vie [TeLink Flasher ](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)upgraded to Zigbee, currently on **LYWSD03MMC-z**,  version  **3001-0127**

> [
    {
        "fileVersion": 2576953345,
        "fileSize": 86386,
        "manufacturerCode": 56085,
        "imageType": 515,
        "sha512": "f663c287b6fa48f2f4df2fae2ff69ca6380c80b32d86a0f92b0c6aa42cb4be98cf9fb4d2f39be22eb29d6354f838f774fb9da2cb2bbc5a0f84e0f8fc8b5e9941",
        "url": "https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/db15-0203-99993001-ATC_v52.zigbee"
    }
]

trablin45, 2025-05-24:
> Just in case someone would be searching for solution OTA on Zigbee2MQTT... this JSON below worked to revert to ATC BLE for LYWSD03MMC (v [3001-100](https://github.com/devbis/z03mmc/releases/tag/1.1.0))), then via [TeLink Flasher ](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)upgraded to Zigbee, currently on **LYWSD03MMC-z**, version **3001-0127**
> 
> > [
> > {
> > "fileVersion": 2576953345,
> > "fileSize": 86386,
> > "manufacturerCode": 56085,
> > "imageType": 515,
> > "sha512": "f663c287b6fa48f2f4df2fae2ff69ca6380c80b32d86a0f92b0c6aa42cb4be98cf9fb4d2f39be22eb29d6354f838f774fb9da2cb2bbc5a0f84e0f8fc8b5e9941",
> > "url": "https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/db15-0203-99993001-ATC_v52.zigbee"
> > }
> > ]

Im trying to revert from 3001-1100 back to BLE and It wont take my zigbee file manually, I tried your json and no updates are found even using the "force" flag. Any other method?