devbis/z03mmc #142: Cannot discover Zigbee after flashing

vad0s94, 2024-04-18:
[Flashing by OTA](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-over-the-air-easy-way) successfully finished, but the device is not discoverable by zigbee coordinator. Reinstalling the battery and RES + GND short have no effect

devbis, 2024-04-23:
Maybe it is still in bluetooth mode? Difficult to tell, but you can always use soldering to write zigbee or bluetooth firmware

Kater-auf-Dach, 2024-07-22:
Have a same issue too. The firmware was flashed whihout errors step by step. However, Zigbee2mqtt doesn't discover device after reinstalling the battery and RES + GND short both

jgdf42, 2024-11-20:
I'm experiencing the same issue. flash z03mmc.bin via uart, device never shows in zigbee coordinator as discoverable. tried erasing nand beforehand, too. pressing reset button for one second does not get the display to switch to Fahrenheit as the notes suggest it should.

when i wipe and flash atc_ota_40000.bin via uart, device shows up in web flasher. from there, flashing z03mmc.bin results in the same thing; no zigbee discoverability, reset button doesn't toggle to fahrenheit.

i may try pvvx's firmware to see if it yields different results.

alinelena, 2025-01-23:
maybe this helps https://github.com/devbis/z03mmc/issues/170#issuecomment-2610934980