devbis/z03mmc #143: Can't add to HA zigbee2mqtt after successful Flash

owlen, 2024-04-27:
I'm hoping to be able to make this work:

I need help adding two flashed devices to my Hame Assitant Zigbee2MQTT addon.

I successfully flashed two "Mi temperature and humidity monitor 2" with this:
https://github.com/atc1441/ATC_MiThermometer/releases/tag/79
Starting up it shows "Atc", and I can control both with the "Telink Flasher for Mi Thermostat" page on my mobile.

I've update the HAOS configuration.yaml file with the `external_converters`:
```yaml
homeassistant: true
external_converters:
  - lywsd03mmc.js
advanced:
  network_key:
    - 151
...
```

and created `lywsd03mmc.js` in the zigbee2mqtt folder, and copied the new version from [converters/lywsd03mmc.js](https://github.com/devbis/z03mmc/blob/96edb6b00166cbbe8bf24a22619cf8188ad343a0/converters/lywsd03mmc.js) in this repo.

My Home Assistant Add On says: `Zigbee2MQTT Current version: 1.36.1-1`.

On the a2mqtt UI I "Permit Join", but the devices do not appear. I tried restarting by removing the battery and waiting a few seconds, as well as resetting by linking the the Gnd and Reset connectors. Nothing appears to try to connect. There is nothing related in the z2mqtt logs either.

Am I missing a step? Is there anything I can do to debug this?

Jonoox, 2024-04-28:
Holding the gnd and reset for around 10 seconds made the connection for me on two thermometers. Fresh Z2M install, without any additional .js files.

owlen, 2024-04-29:
I was able to solve it!

I relied on too many sources, but it worked when I followed the instructions [here](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-over-the-air-easy-way). 
**An important note:**
I used Chrome on both my PC and mobile. I could connect one sensor only to my PC and the other to my mobile. This is very confusing and takes some time to figure out. If others had a similar experience, it's worth mentioning in the doc.

closed 2024-04-29 by owlen