devbis/z03mmc #144: Temperature is reported once every 4 hours. What's wrong?

zverek06, 2024-04-29:
Good afternoon.
Passed the zigbee sensor.
Connected to the Yandex Station 2 hub.

Humidity and charge are transferred normally.

Temperature is reported once every 4 hours. What's wrong?
[image]
[image]
[image]

Roycinger, 2024-05-30:
My Mijia Temp Sensor, flashed with Zigbee FW, reported after first pairing and then never again

leehadassin, 2024-08-21:
Same here - sonoff dongle on homeassistant on rpi5, paired fine, humidity and battery seem to be reporting fine, but temperature hasn’t updated since the initial pair

Coldness00, 2024-12-29:
Same issue.
I need to manually go to Z2M UI and ask for a refresh.