devbis/z03mmc #145: Crashed sensors

ivannunyadambiz, 2024-05-04:
I have the most recent fw provided here, and 3 of the 8 sensors i converted to zigbee keep crashing.
They will not refresh the display anymore and stop transmitting data.

zzz676, 2024-05-05:
> I have the most recent fw provided here, and 3 of the 8 sensors i converted to zigbee keep crashing. They will not refresh the display anymore and stop transmitting data.

Подтверждаю , такая же проблема. Помогло вытащить и вставить батарею.

bbreton09, 2024-05-05:
I have the same issue. The only way to recovery is to remove battery.

zzz676, 2024-05-06:
Датчики продолжают иногда зависать - не обновляется экран, не передается информация по Зигби. Помогает передернуть батарею и тогда сразу актуальная температура на экране появляется.
@devbis

ProTofik, 2024-05-30:
Same here. I had them running rock solid on ATC firmware for the past 2 years but converted them to zigbee about a month or two ago. Since I converted them to zigbee, I had 2 crashes so far out of 9 devices. 

[image]
[image]

KGS501, 2024-05-31:
Same issue here

KGS501, 2024-05-31:
The same problem is discussed in another issue. It seems that there already is a possible fix:

https://github.com/devbis/z03mmc/issues/71