devbis/z03mmc #150: Flash via UART?

RaW79, 2024-05-14:
Hi
I have a LYWSD03MC where something OTA has gone wrong and I´m not sure what is working in the device.

I cannot recognize any BLE activities at different computers but also at different versions of the Online-Flasher.

Then checked my Zigbee cordinator and this also does not recognise any new device within pairing mode like the other LYWSD03MC have flashed successfully before.

So I tried to flash vie the browser UART flasher and connected 3,3V, GND and TX through a 1kOhm resistor. 
Then used z03mmc.bin but of course only with a TX connection there is nothing recognized in the log of the flasher then just the cmd sending to flash.

Afterwards powering the device via battery the h&t values are shown at the display but again no recognition at BLE or Zigbee.

Can someone please give  some more details how to do the flashing?

I also shorted GND and RST points but only due to longer connection I can change display values from °C to F.

Kindly ask for your guidance. Thx!

RaW79, 2024-05-28:
Thanks for your feedback and I´ll try the python script.
To be sure regarding the readme and connection [Mi_SWS_Connection.jpg](https://github.com/devbis/z03mmc/blob/master/assets/Mi_SWS_Connection.jpg) What is to be connected at P14?  Both TX (with resistor) and RX (direct) at the same time? Sorry for asking but new for me and never connected RX/TX at the same pin (only 3 wires in the picture) and because it does not work so far it´s not clear in the picture.

devbis, 2024-05-28:
Yes, both to P14. It is an emulation of single wire protocol by using a resistor to distinguish 'in' and 'out' signals.

c15yi, 2025-01-15:
Hello, I am struggling to flash via UART. I tried two different USB to TTL/UART adapters, but none of them seems to work with the script.

The command I used was this `python3 z03mmc/TLSR825xComFlasher.py -p /dev/tty.SLAB_USBtoUART -d -t100 -b 921600 wf 0 ATC_ota_40000.bin`

The first one is this one: https://www.amazon.de/dp/B09SZ3VDN2?ref=ppx_yo2ov_dt_b_fed_asin_title
<details>
<summary>Output Waveshare Adapter</summary>

```
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/tty.usbmodem575C0039201, 921600 baud...
Reset module (RTS low)...
Activate (100 ms)...
Debug: default swdiv for 24 MHz = 52 (0x34)
Debug (read data):
000023: fe 
Debug (read data):
000024: fe 
Debug (read data):
000025: fe 
Debug (read data):
000026: fe 
Debug (read data):
000027: fe 
Debug (read data):
000028: fe 
Debug (read data):
000029: fe 
Debug (read data):
00002a: fe 
Debug (read data):
00002b: fe 
Debug (read data):
00002c: fe 
Debug (read data):
00002d: fe 
Debug (read data):
00002e: fe 
Debug (read data):
00002f: fe 
Debug (read data):
000030: fe 
Debug (read data):
000031: fe 
Debug (read data):
000032: fe 
Debug (read data):
000033: fe 
Debug (read data):
000034: Debug (read data):
000035: fe 
Debug (read data):
000036: fe 
Debug (read data):
000037: fe 
Debug (read data):
000038: fe 
Debug (read data):
000039: fe 
Debug (read data):
00003a: fe 
Debug (read data):
00003b: fe 
Debug (read data):
00003c: fe 
Debug (read data):
00003d: fe 
Debug (read data):
00003e: fe 
Debug (read data):
00003f: fe 
Debug (read data):
000040: fe 
Debug (read data):
000041: fe 
Debug (read data):
000042: fe 
Debug (read data):
000043: fe 
Debug (read data):
000044: fe 
Debug (read data):
000045: fe 
Debug (read data):
000046: fe 
Debug (read data):
000047: fe 
Debug (read data):
000048: fe 
Debug (read data):
000049: fe 
Debug (read data):
00004a: fe 
Debug (read data):
00004b: fe 
Debug (read data):
00004c: fe 
Debug (read data):
00004d: fe 
Debug (read data):
00004e: fe 
Debug (read data):
00004f: fe 
Debug (read data):
000050: fe 
Debug (read data):
000051: fe 
Debug (read data):
000052: fe 
Debug (read data):
000053: fe 
Debug (read data):
000054: fe 
Debug (read data):
000055: fe 
Debug (read data):
000056: fe 
Debug (read data):
000057: fe 
Debug (read data):
000058: fe 
Debug (read data):
000059: fe 
Debug (read data):
00005a: fe 
Debug (read data):
00005b: fe 
Debug (read data):
00005c: fe 
Debug (read data):
00005d: fe 
Debug (read data):
00005e: fe 
Debug (read data):
00005f: fe 
Debug (read data):
000060: fe 
Debug (read data):
000061: fe 
Debug (read data):
000062: fe 
Debug (read data):
000063: fe 
Debug (read data):
000064: fe 
Debug (read data):
000065: fe 
Debug (read data):
000066: fe 
Debug (read data):
000067: fe 
Debug (read data):
000068: fe 
CLK == 0. Chip sleep? -> Use reset chip (RTS-RST): see option --tact
```
</details>

The other one is this one: https://www.amazon.de/dp/B07XZ3PV8T?ref=ppx_yo2ov_dt_b_fed_asin_title
<details>
<summary>Output Hailage Adapter</summary>

```
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/tty.SLAB_USBtoUART, 921600 baud...
Reset module (RTS low)...
Activate (100 ms)...
Debug: default swdiv for 24 MHz = 52 (0x34)
Debug (read data):
000023: Debug (read data):
000024: Debug (read data):
000025: Debug (read data):
000026: Debug (read data):
000027: Debug (read data):
000028: Debug (read data):
000029: Debug (read data):
00002a: Debug (read data):
00002b: Debug (read data):
00002c: Debug (read data):
00002d: Debug (read data):
00002e: Debug (read data):
00002f: Debug (read data):
000030: Debug (read data):
000031: Debug (read data):
000032: Debug (read data):
000033: Debug (read data):
000034: Debug (read data):
000035: Debug (read data):
000036: Debug (read data):
000037: Debug (read data):
000038: Debug (read data):
000039: Debug (read data):
00003a: Debug (read data):
00003b: Debug (read data):
00003c: Debug (read data):
00003d: Debug (read data):
00003e: Debug (read data):
00003f: Debug (read data):
000040: Debug (read data):
000041: Debug (read data):
000042: Debug (read data):
000043: Debug (read data):
000044: Debug (read data):
000045: Debug (read data):
000046: Debug (read data):
000047: Debug (read data):
000048: Debug (read data):
000049: Debug (read data):
00004a: Debug (read data):
00004b: Debug (read data):
00004c: Debug (read data):
00004d: Debug (read data):
00004e: Debug (read data):
00004f: Debug (read data):
000050: Debug (read data):
000051: Debug (read data):
000052: Debug (read data):
000053: Debug (read data):
000054: Debug (read data):
000055: Debug (read data):
000056: Debug (read data):
000057: Debug (read data):
000058: Debug (read data):
000059: Debug (read data):
00005a: Debug (read data):
00005b: Debug (read data):
00005c: Debug (read data):
00005d: Debug (read data):
00005e: Debug (read data):
00005f: Debug (read data):
000060: Debug (read data):
000061: Debug (read data):
000062: Debug (read data):
000063: Debug (read data):
000064: Debug (read data):
000065: Debug (read data):
000066: Debug (read data):
000067: Debug (read data):
000068: CLK == 0. Chip sleep? -> Use reset chip (RTS-RST): see option --tact
```
</details>

What could be the issue here? I also tried without the `-b` options, but I get the same results.

devbis, 2025-01-16:
There is a chain of actions. When device is working it uses SWS pin as gpio and it is difficult to flash it.

For this case you need to 

- turn off the device (but not UART)
- run the command with -t5000 (5 seconds). It will send starting packets for 5 seconds. 
- turn the device again in 5 seconds. It should be running in flashing mode.

If 5 seconds is not enough, you can increase the 'tact' variable.

RaW79, 2025-01-16:
For me it was also important to connect RST to the adapter. In the manual or the browser version of the flasher it´s marked as option. Guess for a reboot with the right timing this helps a lot and is only working method for me.

c15yi, 2025-01-16:
I actually used the RST pin for starting/resetting the device. That's why I have the option `-t 100` specified in the command. I saw that the code will trigger the RST in the activation if a time is set via that option. 

Are there ways to verify that the devices are not bricked completely? I already erased the memory (I think, it's not showing up Bluetooth discovery). 

Are the adapters I have the correct ones? Both are for UART but they yield different results?!

AussiSG, 2026-02-25:
@devbis thank you for the reply.
The sensor is currently equipped with zigbee firmware. 
Not really happy with that so want to revert to BLE again.

Will give this a shot tonight!

AussiSG, 2026-02-25:
Hmmm I was able to perfom some OTA updates through Zigbee2Mqtt. 
They succeeded but now I am unable to find the sensor with any tool. 
Not in Home Assistant or via https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

I First update to zigbee
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/devbis/1141-020a-09993001-z03mmc_v1100.zigbee

Then updated to Bluetooth
https://github.com/devbis/z03mmc/blob/master/assets/db15-0203-99993001-ATC_v46.zigbee