devbis/z03mmc #153: Device showing up weird cluster endpoints on ZHA

Mr-Alicates, 2024-05-25:
Hello, 
I'm using the latest firmware (0x11003001) on HA 2024.5.5 with ZHA

When trying to use the attributes of the User interface cluster, I see this attributes:

[image]

Which do not match the description provided in the repository.

However, I've been able to successfully use them with ZHA_Toolkit.

Also the thermometers show this entities when paired, and I think there might be something wrong:
[image]

devbis, 2024-05-27:
Yeah, those parameters looks weird, but they are required for thermostat UI cluster that I'm using to change Cº/Fº. You can disable these entities freely.

gigamonster256, 2024-08-28:
I'm observing the same, no mention of the new customization attributes in the ZHA interface or diagnostic output.

I've never worked on Zigbee stuff before and it looks like [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/custom_devices_diy.ts#L761) is where the device is defined for zigbee2mqtt, but I can't find any mention of the device in zigpy (what ZHA uses.) It looks like a special [device handler](https://github.com/zigpy/zha-device-handlers) may need to be written for the nonstandard customization attributes to work in ZHA.

gigamonster256, 2024-08-28:
Here's a ZHA quirk that fixes the issue you're seeing and exposes the display configuration attributes in the zigbee management interface: https://github.com/zigpy/zha-device-handlers/blob/d864e35e661824a65c5714cded18b4d7cf75c2fe/zhaquirks/xiaomi/custom/z03mmc.py. You can enable it by following the instructions here: https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226 and I am hoping to get it upstreamed into the main zha-device-handlers repo.