devbis/z03mmc #154: zigbee change offsets of humidity and temp?

RMHogervorst, 2024-06-21:
I feel this is a dumb question, but can I modify the temperature and humidity sensor offsets through zigbee?

I flashed a device and realise the temp is off.

RMHogervorst, 2024-06-22:
Preferably through home assistant

devbis, 2024-06-22:
Yes you can. https://github.com/devbis/z03mmc?tab=readme-ov-file#configuring-the-device
You can write corresponding values to the mentioned clusters/attributes for the result you need.

RMHogervorst, 2024-06-25:
Okay I've tried several things but I don't know how to change the values through home assistant.

Could you help me? 
Here is what I tried:

## Modify values through the device page

- Go to device page
- click on three dots under device info. 
- click on manage zigbee device

In there we have values for temperature measurement 0x0402. and a min, a max and some more attributes, but not attribute: 0x0010.

I can write values but that doesn't seem to have an effect.

## use developer tools / call service / issue zigbee cluster command

- go to developer tools
- go to tab call service
- pick issue zigbee cluster command

I know the ieee of the device, I know the cluster id (1) but I am a bit lost on what to do next.
Do you have an example yaml that I can use?

RMHogervorst, 2024-06-25:
oh noo, do I need to translate the hex to integer?

- cluster_id: 0x0402 = 1026
- attribute: 0x0010 = 16

This at least gives me a green check mark on execution. 

```yaml
service: zha.set_zigbee_cluster_attribute
data:
  cluster_type: in
  ieee: [mac address]
  cluster_id: 1026
  endpoint_id: 1
  attribute: 16
  value: -1000
```

But this does not seem to change the values I get back from the device

nsteenbeek, 2024-09-20:
Same here. Tried to turn off the smiley:

* cluster_id: 0x0204 = 516
* attribute: 0x0010 = 16
* value: 0

Also greenmark on execution, but no change. 
Changing to Fahrenheit/Celcius works when using attribute: 0x0000 = 0