devbis/z03mmc #156: Device stop reporting since july 2nd

sybux, 2024-07-17:
Hello,

i've upgrade 12 ATC device with your zigbee firmware perfectly for about 4 months now.
Suddenly, on of my device stop reporting. The value is always 24.11° and 63.11% for humidity.
On the device it self, display is stuck with value. If I remove battery, value change once, data are never sent to Z2M nor HA.

Any other like this ? What can I provide more to help debuging ?

Didel, 2024-07-18:
I have the same issue (but with different percentages). Everything was working fine in HA, but on July 10th, it suddenly stops reporting new values. I tried to reflash the device, but the browser can't fine the device when using bluetooth to scan for it.

sybux, 2024-07-18:
I've finally find my problem ... a stupid one ! Battery is empty (3% of charge). So it can display data but can't update, nor transmit to Z2M :(....
Just added a monitoring on all battery in my HA device do avoid further problem of this kind !

Didel, 2024-07-19:
Hhm. A new battery has also resolved my issue... Could it be that reading or reporting the battery percentage is off? Perhaps something that @devbis could have a look at?

sybux, 2024-07-19:
I don't think nor I'm sure that nothing can be done at the firmware level. You need to add a control in HA to display battery level. As if battery is dead, connection cannot be made so report cannot be done. see bellow, my battery dashboard as example :
[image]

Didel, 2024-07-19:
@sybux I am very familiar with showing the battery percentage of the device in HA :) As we both experienced, the thermometer had enough battery to keep running the display. I know that the device has a 'battery low' icon (below the smiley), but that was not visible while experiencing the problems. Also, the last reported battery percentage in HA was 43%, which I did not consider 'low' and in my opinion should be enough to power the device with all its features. My assumption is that the reported battery percentage is different (higher) from the actual usable battery percentage (e.g. due to firmware doing wrong readings or calculations). It seems like the remaining percentage was enough to keep the display running, but not enough to do new measurements or power the bluetooth radio to report readings to HA.

I would suggest to map the reported battery percentage to a range where it would still be able to report bluetooth readings, and if e.g. the battery falls below "10%" usable , it should only show the remaining battery percentage and the battery icon on the actual device. Note that "10% usable" might in reality mean that there is still more than 50% remaining current in the battery, but that would not be enough to power both the thermometer and bluetooth circuits.

sybux, 2024-07-19:
from my side, last report for battery was 5%

Christian-PK, 2024-07-21:
I am experiencing the same issue. I Just changed the battery to check if it makes any difference. I moved the old battery to a Ikea button. The xiaomi temperature sensor reported 63% and the ikea button reported 60 % so i dont think the battery percentage is wrong.

vitosrnd, 2024-07-23:
I have 3 of 3 devices hangs with almost full battery. It seems that the device hangs at the moment of packet sending. Maybe battery can't hold peak current and voltage drops down for a short time. I want to power it from an external 18650 Li-ion battery, and later I'll write about result.

pauliusnar, 2024-07-28:
I have 6 thermometers and at least 3 of them crashes - LCD screen freezes with last results, but home assistant keeps working normally
[image]
[image]

LCD keeps showing same temperature and same humidity. If battery is removed and inserted again - lcd starts to show real measurements. 
Interesting point - while device is crashed it reports same humidity value
but once the battery removed and re-inserted it starts to work normally
[image]

pauliusnar, 2024-07-28:
I did not change the battery, just re inserted same one. Reset every 24h would help and would be easy to implement.

Christian-PK, 2024-07-28:
After battery change it has worked without issues for a week now. But wierd it does not work properly with 60%+ battery. As mentioned in early comment i verified the battery percentage by moving the battery to a hue device and it also reported 60%+ battery.

pauliusnar, 2024-08-12:
Continued. After moving my thermomers next to coordinator, probably it fixes the problem and the thermometer shows correct values. Try placing it near transmiter, may be helpful to troubleshoot

dziedzic99, 2024-09-03:
Can confirm I have a same problem and it's low battery. Is it so that this firmware drains the battery faster? Maybe there is a way to improve this? Any tips?

frickua, 2024-09-04:
Some freeze with several sensors. I will check the new batteries.

HA show battery percentages based on voltage, but if battery has low quality(capacity\ampers characteristics) it can be inaccurate. And can not handle high current.

ProTofik, 2024-09-05:
Same issue on all 9 of my sensors. The screen gets stuck and it stops reporting when the battery is low. I measured the 'dead' battery and it's 3.03V which in my opinion should be plenty enough.

Soukyuu, 2024-09-10:
Half of my sensor got stuck yesterday as well. It's not even a dead battery, because re-inserting the battery *while shorting the reset pins* brings them back to life - for a while at least. My zigbee network is really dense, over 100 devices, so reception shouldn't be a problem either.

It's a great achievement to be able to run zigbee on the bt hardware, but for me, it's not stable enough for the upcoming heating season. I'm going to keep an eye on this repo, but for now I'm back to bluetooth.

frickua, 2024-09-10:
> Same issue on all 9 of my sensors. The screen gets stuck and it stops reporting when the battery is low. I measured the 'dead' battery and it's 3.03V which in my opinion should be plenty enough.

Voltage characteristic isn't enough. Bad batteries can have good voltage but can not produce enough current to transmit data.
@Soukyuu I think when sensor frize, the battery can recover a little, but in my experience reset helps only for several hours.