devbis/z03mmc #160: How to OTA custom firmware?

aurbod, 2024-08-28:
Hello! This is a follow-up question to issue #18 : I would like to create a custom firmware and upload it via OTA. So far, I have not been successfull. This is what I tried:
- I made my changes (**g_zb_txPowerSet** in _src/common/main.c_)
- I changed _src/version_cfg.h_ as follows:
  - APP_RELEASE from 0x11 to 0x12
  - APP_BUILD from 0x00 to 0x01
  - MANUFACTURER_CODE_TELINK from 0xdb15 to 0xdb16
- I then built the firmware according to the README:
```
cmake -B build -DSDK_PREFIX=$(pwd)/../tl_zigbee_sdk -DTOOLCHAIN_PREFIX=$(pwd)/../tc32 -DMANUFACTURER_CODE=0x1141
cmake --build build --target z03mmc.zigbee
```
I ended up with a file _1141-0203-12013001-z03mmc.zigbee_ which I copied to my zigbee2mqtt OTA directory and I updated _ota.json_ as follows:
```
[
    {
        "force": true,
        "url": "/opt/zigbee2mqtt/data/ota/1141-0203-12013001-z03mmc.zigbee"
    }
]
```

However, it always says "No update available":
```

info  2024-08-28 06:26:06: Checking if update available for '0xa4c138631a34bea1'
info  2024-08-28 06:26:12: No update available for '0xa4c138631a34bea1'
info  2024-08-28 06:26:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138631a34bea1', payload '{"battery":100,"comfort_humidity_max":85,"comfort_humidity_min":50,"comfort_temperature_max":25,"comfort_temperature_min":20,"enable_display":"ON","humidity":56.64,"humidity_calibration":0,"linkquality":160,"show_smiley":"SHOW","temperature":23.42,"temperature_calibration":0.04,"temperature_display_mode":"celsius","update":{"installed_version":285224961,"latest_version":285224961,"state":"idle"}}'
info  2024-08-28 06:26:12: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0xa4c138631a34bea1","updateAvailable":false},"status":"ok","transaction":"mf831-1"}'
```

I even tried the add.js script from https://github.com/Koenkk/zigbee-OTA/tree/master to create my very own index.json, but this did not work either. Am I missing something? Do I need to change the "-DMANUFACTURER_CODE" in the build command?

aurbod, 2024-08-28:
For reference this is my index.json created by add.js

```
cat index.json
[
    {
        "fileVersion": 302067713,
        "fileSize": 131378,
        "manufacturerCode": 4417,
        "imageType": 515,
        "sha512": "ae83e077d91fc4b2ca658c2d9754daa6675b632863f62aa1edd786bf909b2541c25b63752972b791be546527cecfca99c50c598085c3a948e72a206a95b99444",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Telink/1141-0203-12013001-z03mmc.zigbee",
        "path": "images/Telink/1141-0203-12013001-z03mmc.zigbee"
    }
]
```

I've changed the url to _/opt/zigbee2mqtt/data/ota/1141-0203-12013001-z03mmc.zigbee_ and removed the "path" entry, this did not work either.

Evul88, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

devbis, 2024-08-28:
Before starting OTA the device checks OTA file's header against manufacturer ID and device ID. 
In 1.1.0 vendor ID is 0xdb15 and device ID is 0x0203
You need to keep these values OR mangle them in OTA header. 
I just added options in tools/make_ota.py to let you set old values in the header and use your own ones in the binary.

aurbod, 2024-08-28:
Hi Ivan, thank you so much for your quick reply and fix! I was able to update my custom firmware as follows:

- made my changes (**g_zb_txPowerSet **)
- Changed only **APP_BUILD** from 0x00 to 0x01 in _src/version_cfg.h_
- Compiled the binary as instructed in the README (see above)
- Then modified the file:
```
cd build/src/
python3 ../../tools/make_ota.py -m 0xdb15 -t 0x0203 z03mmc.bin
# This produced a file called db15-0203-11013001-z03mmc.zigbee
cp db15-0203-11013001-z03mmc.zigbee /opt/zigbee2mqtt/data/ota/
cat /opt/zigbee2mqtt/data/ota.json
[
    {
        "force": true,
        "url": "/opt/zigbee2mqtt/data/ota/db15-0203-11013001-z03mmc.zigbee"
    }
]
# Restarted zigbee2mqtt
```

That was it! Afterwards I did a manual update check in zigbee2mqtt, it detected the new firmware, the whole OTA took around 15 min. I've posted the result in issue #18 

Thanks!

closed 2024-08-28 by aurbod