devbis/z03mmc #162: Zigbee OTA to BLE not working

HennieLP, 2024-10-01:
if i add the Zigbee [db15-0203-99993001-ATC_v48.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/db15-0203-99993001-ATC_v48.zigbee) file to my ota folder and rename it to .ota it just says Newes version installed if i try to update it to the BLE Version

jksmurf, 2024-10-14:
According to the procedure you don’t rename .zigbee to .ota.

see https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/

having said that the procedure doesn’t work for me it just keeps saying I have the latest FW installed.

nobwyn, 2024-10-14:
I have the same issue - Zigbee2Mqtt shows me OTA update with v48 (tried v47 and v46 as well), but once I try to do the update - it say newest version already installed... 

So how to revert back to BLE?

jksmurf, 2024-10-14:
I got it working with pvvx V47 in the end. 

What got it going was a HA restart and not just a Zigbee2MQTT restart.

derbullifahrer, 2025-01-18:
Hi everyone,
I try to revert back to BLE from Version 3001-1100. OTA update always say: "not a valid OTA file". 
I tried it with three different firmware files: 
1141-0203-99993001-ATC_v46.zigbee
1141-020a-01233001-Z03MMC.zigbee
db15-0203-99993001-ATC_v46.zigbee

Anyone an idea?

ezcGman, 2025-01-23:
Did somebody ever achieve rolling back to BLE using ZHA, as the inital poster asked? ZHA is really a pain in the ass if it comes to this...

jksmurf, 2025-01-23:
> Did somebody ever achieve rolling back to BLE using ZHA, as the inital poster asked? ZHA is really a pain in the ass if it comes to this...

As above; yes.

ezcGman, 2025-01-23:
> > Did somebody ever achieve rolling back to BLE using ZHA, as the inital poster asked? ZHA is really a pain in the ass if it comes to this...
> 
> As above; yes.

!? Where? You talk about Zigbee2MQTT, not ZHA

jksmurf, 2025-01-23:
> > > Did somebody ever achieve rolling back to BLE using ZHA, as the inital poster asked? ZHA is really a pain in the ass if it comes to this...
> > 
> > 
> > As above; yes.
> 
> !? Where? You talk about Zigbee2MQTT, not ZHA

Oops sorry, it’s been a while, yes, mine was Zigbee2MQTT back to BLE.

arnizzz, 2025-02-17:
Zigbee2MQTT 1.42.0-1 LYWSD03MMC-z Firmware version 3001-0124 trying back to BLE.
I think all config like my_index.json bits done correctly I am getting error no file or directory, but after uploading 1141-0203-99993001-ATC_v46.zigbee to ota folder all the time respond is "z2m: no update available for..."

derbullifahrer, 2025-02-18:
Zigbee2mqtt shows Custom devices (DiY) - LYWSD03MMC with Firmware-Version 3001-1100
Zigbee log: installed_version":285224961,"latest_version":285224961

whitch firmware file is to use to get device updated to LYWSD03MMC**-z** with Firmware 3001-0123 ?

arnizzz, 2025-02-18:
I used the file Zigbee Firmware: Z03MMC_v0124.bin assigned in pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html 
Thank you Devbis.  I understood that LYWSD03MMC is 0203 but after trying a different 020a from pvvx/ATC_MiThermometer, I got a good update. Now it is in BLE.

derbullifahrer, 2025-02-18:
@arnizzz whitch one did you use? can you post link and filename? would be helpful. - Thanks

arnizzz, 2025-02-18:
pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v50.zigbee

derbullifahrer, 2025-02-18:
still doesn't get OTA to run. Error is "z2m: Failed to check if update available for 'Xiaomi' (Not a valid OTA file)"

mcbyte-it, 2025-03-14:
Same issue here, currently on:
`Custom devices (DiY) | LYWSD03MMC | 20231109 | 3001-1006`

Can't update to any other firmware, not even newer zigbee, always give no update available.
I tried with the 0203 and 020a versions, v46-v50-v51. always the same.

I have a SONOFF ZBDongle-E with firmware v7.4.4