devbis/z03mmc #164: Temp module disappear from my ZHA

ranrinc, 2024-10-06:
I'm trying to recover my zigbee temp back after it become unavailable from my zigbee network. However I cant seem to be able to reset it. trying to connect reset + GND for 5 sec, seems to have no luck. Anyone have idea on the best way to reset the device? What will the screen shown/changes on discover mode?

Please help. Thanks

dsimoes, 2024-10-10:
Also with the same problem

dsimoes, 2024-10-10:
Nevermind. @ranrinc Make sure the battery is new, a low battery might show the display but not have enought power for zigbee functions.

Also, the reset is a bit annoying.. Try to remove the battery, hold the two pins. Then re-insert the battery.