devbis/z03mmc #165: ZigBee on display

ANTPro, 2024-10-14:
Перестал передавать данные в z2m. Перекинул батарею и девайс вошёл в режим:
[image]
Попробовал на другом девайсе и тоже самое, но данные приходят. Что за режим такой и как починить? Версия прошивки 1.1.0

Ibergard, 2024-10-18:
у меня такая же история, только без режима отключенного дисплея.

показания на экране не меняются и при передергивании батарейки сначала отображается то же самое, что на приложенном выше фото, а потом те же не изменяющиеся показания, что и до перезагрузки.

kahatie, 2024-12-10:
hi 
i get have same 
i disabled display (for save battery) but it stop send update to z3m and display stay on 

i try remove battery for reboot it but now only display this

Allineer, 2024-12-20:
Села батарейка, поменял через какое-то время. На экране то же самое, что в первом сообщении, и экран не выключается какими-либо настройками со стороны z2m.

devbis, 2024-12-20:
https://github.com/devbis/z03mmc/blob/master/src/device.c#L165

It is the initial message on the display, during normal work it is almost not visible - temperature and humidity starts to display immediately after it. 

But in case of problems (e.g. low battery) it might get stuck. 
I don't have a solution for this case besides changing the battery.

Allineer, 2024-12-20:
> I don't have a solution for this case besides changing the battery.
I tried.

devbis, 2024-12-20:
In this case you may try to use wires (as it's written in the README): erase the chip and flash the firmware again. 
Mac address will be changed and you'll need to join it again to the network.

Allineer, 2024-12-20:
Thank you. I'll try someday ... when wires appear)