devbis/z03mmc #168: Activating takes forever

rednasss, 2024-11-12:
Hello,
How long should activating take once clicked on "Do Activation"?
I am now already waiting for more than 10 minutes and it keeps saying: "Activating now, please wait..."
Thanks!

devbis, 2024-11-13:
I guess, you have a new firmware 2.1.1. It is described in readme. You need to use wires to flash the device for the first time

closed 2024-11-15 by devbis

rednasss, 2024-11-16:
Do you know if there will be a possibility to do OTA  flashing with this firmware anywhere in the near future?

devbis, 2024-11-17:
I doubt it. It is easier to use wires to flash any firmware, ble or zigbee for the first time.
To track unsupported firmware you can check here https://github.com/atc1441/ATC_MiThermometer/issues/298

rednasss, 2024-11-17:
That's sad, then I won't be able to do it unfortunately..
Hope there will be a fix ever