devbis/z03mmc #169: Receiving the an eoor when i try and flash the zigbee firmware OTA

tigger29900, 2024-11-21:
Had no issues flashing custom firmware version 4.7 when using https://devbis.github.io/telink-zigbee/ but when I try and flash the z03mmc.bin I get the below error it is hardware LYWSD03MMC HW:B1.5

1:39:02 PM: Update error: NotSupportedError: GATT operation failed for unknown reason.

ASergeyV, 2024-11-24:
First, flash ATC_ota_40000.bin from this page and not custom firmware version 4.

devbis, 2024-11-25:
As far as I remember pvvx's 4.7 had an issue with OTA upgrading for rev 1.5/1.6
It was fixed in 4.8 and later versions. The only way now is to flash using wires. Any of zigbee/ble firmware.

ASergeyV, 2024-11-25:
Yesterday I flashed from 1.5, first flashed firmware version 4.7, got the same error, then flashed from this site ATC_ota_400000 and then flashed z03mmc.bin. Everything is via OTA. Read the instructions https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-over-the-air-easy-way

tigger29900, 2024-11-25:
Used the ATC_ firmware and it worked flawlessly

Now I just need to revive the two I bricked