devbis/z03mmc #171: Interview failed because can not get active endpoints

dabepp, 2024-11-24:
I updated like 8 or 9 Xiamo LYWSD03MMC to Zigbee FW Z03MMC_v0123.bin and they work flawless, but one somehow has issues.
I did do a reset, first it showed Farenheit, another reset with the battery in switched it to Celsius, so at least it is not totally bricked...

This is the message the log shows, search did not find anything.

`info 2024-11-24 22:28:58zh:controller: Interview for '0xa4c1388b65c5f629' started
info 2024-11-24 22:28:58z2m: Device '0xa4c1388b65c5f629' joined
info 2024-11-24 22:28:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c1388b65c5f629"},"type":"device_connected"}'
info 2024-11-24 22:28:58z2m: Starting interview of '0xa4c1388b65c5f629'
info 2024-11-24 22:28:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c1388b65c5f629"},"type":"pairing"}'
info 2024-11-24 22:28:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388b65c5f629","ieee_address":"0xa4c1388b65c5f629"},"type":"device_joined"}'
info 2024-11-24 22:28:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388b65c5f629","ieee_address":"0xa4c1388b65c5f629","status":"started"},"type":"device_interview"}'
error 2024-11-24 22:29:09zh:controller:device: Handling of poll check-in from 0xa4c1388b65c5f629 failed (ZCL command 0xa4c1388b65c5f629/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (no response received (130) Error: waiting for response TIMEOUT))
error 2024-11-24 22:29:09zh:controller:device: Handling of poll check-in from 0xa4c1388b65c5f629 failed (ZCL command 0xa4c1388b65c5f629/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (no response received (129) Error: waiting for response TIMEOUT))`

It is online, though... 

[image]

Anything I can do besides soldering some wires to the board and flash it via serial connection?

devbis, 2024-11-25:
v0123 - this version is from a pvvx's fork.  Not firmware from this repo. You may try to re-interview the sensor using blue or yellow buttons on the bottom left.

dabepp, 2024-11-26:
> did you solve it?

No, there are no buttons as @devbis mentioned on those Xiaomi Thermometers (At least not visible w/o dismantling everything...)

nmaupu, 2025-02-06:
You should probably delete it from z2m, reset it (short reset+ground for ~10 secs) and repair it...