devbis/z03mmc #172: Z2M cannot load converter 'vindriktning.js'

lotohov, 2024-12-01:
Failed to load external converter file 'vindriktning.js' (Cannot read properties of undefined (reading 'options'))
[2024-12-01 00:31:00] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-12-01 00:31:00] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device

<hr /><em>This discussion was created from the release <a href='https://github.com/devbis/z03mmc/releases/tag/'>IKEA Vindrikntning zigbee upgrade</a>.</em>

_Originally posted by @devbis in https://github.com/devbis/z03mmc/discussions/158_

devbis, 2024-12-01:
It looks like your converter is broken. What is the content? What z2m version do you have?

lotohov, 2024-12-01:
Z2m 1.41.0-1 as ha addon i have another working converter and im using you converter from that url. 
Zigbee2mqtt converter: https://raw.githubusercontent.com/devbis/z03mmc/vindriktning/converters/vindriktning.js

devbis, 2024-12-02:
https://raw.githubusercontent.com/devbis/z03mmc/57e427ea172fc5ba58a579ddebd2789bdb27eb13/converters/vindriktning.js

Try the converter before the last commit. It was tested better

lotohov, 2024-12-02:
> https://raw.githubusercontent.com/devbis/z03mmc/57e427ea172fc5ba58a579ddebd2789bdb27eb13/converters/vindriktning.js
> 
> Try the converter before the last commit. It was tested better

this working ! but have commented out pm10 and pm01 ?

lotohov, 2024-12-02:
Thank u for projects and support!

closed 2024-12-03 by devbis