devbis/z03mmc #177: Display not updated

DocEric, 2025-01-11:
First of all: Thank you very much for your work on this!

A few days ago I got my sensors and flashed them immediately. Flashing went well, Zigbee connection went thru without issues, everything seemed to be fine. However, I notice that the display is stuck, meaning it shows always the same (meanwhile incorrect) values as if it's not being updated. However, values are reported via Zigbee just fine. Battery is at 99%, I even changed the battery to a brand new one. 
Disconnecting the battery and inserting it again seems to help for a while, and the values on the display are the same as via Zigbee, but a little later the display is stuck again.

Do you have any idea what could be the problem?

gblaszczuk, 2025-03-16:
I have the same issue.

giuseppe99barchetta, 2025-05-31:
Same problem here. Display won't update values. Zigbee updates correctly.