devbis/z03mmc #179: Does this work with the new exploit?

kcajjones, 2025-01-21:
A new exploit for firmware version 2.1.1_0159 has been discovered here:
[https://github.com/atc1441/ATC_MiThermometer](url)

After flashing the 2.1.1_0159 exploit, can the normal OTA flashing procedure work to get this ZigBee firmware working?

devbis, 2025-01-21:
Yes, it should work. After flashing to ATC firmware with that firmware you can convert it to zigbee according to the instructions

kcajjones, 2025-01-21:
When trying to pair the device via bluetooth on Telink Flasher v6.2, I get the following errors:

> 11:35:34: Searching for devices
> 11:35:37: Connecting to: ATC_A74026
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Reconnect 1 from 5
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Reconnect 2 from 5
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Reconnect 3 from 5
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Reconnect 4 from 5
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Reconnect 5 from 5
> 11:35:40: TypeError: getDevVersion(...).then is not a function
> 11:35:40: Something went wrong, too many reconnects

And when trying to connect for a second time:

> 11:36:53: Searching for devices
> 11:36:56: Connecting to: ATC_A74026
> 11:36:56: Disconnected.
> 11:36:56: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
> 11:36:56: Reconnect 1 from 5
> 11:37:00: TypeError: getDevVersion(...).then is not a function
> 11:37:00: Reconnect 2 from 5
> 11:37:00: TypeError: getDevVersion(...).then is not a function
> 11:37:00: Reconnect 3 from 5
> 11:37:00: TypeError: getDevVersion(...).then is not a function
> 11:37:00: Reconnect 4 from 5
> 11:37:00: TypeError: getDevVersion(...).then is not a function
> 11:37:00: Reconnect 5 from 5
> 11:37:00: TypeError: getDevVersion(...).then is not a function
> 11:37:00: Something went wrong, too many reconnects

kcajjones, 2025-01-21:
I used this web flasher tool successfully from the latest (as of 21/01/25) exploited firmware for 2.1.1_0159 [https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html](url)

nikosece, 2025-01-21:
> I used this web flasher tool successfully from the latest (as of 21/01/25) exploited firmware for 2.1.1_0159 [https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html](url)
> 
> Flashing the ATC_ota_40000.bin file took a long time (~300 seconds). I then removed the battery and re-inserted. I was then able to connect again and flash the z03mmc.bin file. The flash was a lot faster and rebooted into zigbee pairing mode. It was immediately available and connected to my Home Assistant server via ZHA.

The same method worked for me. Just finished!

Fredyy90, 2025-01-29:
I have some new thermometers with firnmware 2.1.1_0159, installing the exploit works fine the ATC firmware works, but after installing z03mmc via OTA they seem bricked, I can only recover them by installing the ATC firmware again via UART.

Installing z03mmc via UART also doesn't help, the don't show anything on the display and also dont seem to send anything on zigbee my ZHA can't find them. Did I get a new hardware revision?

devbis, 2025-01-29:
> but after installing z03mmc via OTA they seem bricked

Did you use transitional ATC_ota_40000.bin before flashing zigbee?

Shuai-Xin, 2025-02-05:
@Fredyy90 Did you manage to solve the problem?

CoolOn3, 2025-02-20:
> I have some new thermometers with firnmware 2.1.1_0159, installing the exploit works fine the ATC firmware works, but after installing z03mmc via OTA they seem bricked, I can only recover them by [...]

Some/similar problem!
In fact you notice the device is working but does not show image.
I can get in the mode to do Join to the Zigbee network (3 seconds GND and RST) and then see the equipment on the network without problem but without image on the display, I tried everything but without luck
This only happens with the last ones I bought, I have 6 other working perfectly. Curious that if you return to flash the firmware ATC (V4 or V5) works all right (via Bluetooth)

Hardware Revision String: B1.5
Software Revision String: 0159
Firmware Revision String: 2.1.1_0159

gblaszczuk, 2025-03-16:
Same problem on two separate devices 2.1.1_0159. One device the screen never turns on while the other device the screen does not refresh. Both devices do show up in zigbee2mqtt and report correct values.

Damianod27, 2025-03-30:
I installed Zigbee software for three Xiaomi sensors. Two had original software version 1.1 and work without a problem in Zigbee2MQTT. In the case of the third, where the software was version 2.1.1_0159, I have a similar case. According to the instructions, I first uploaded the software version ATC_ota_40000.bin via OTA. Then z03mmc.bin. Since then, the display has not refreshed and I cannot pair with Zigbee2MQTT. Bluetooth does not work either. I tried uploading the software from Devbis and PVVX via UART, but the result is the same. Zigbee and Bluetooth do not work and the screen does not refresh. Uploading the original software via UART from Xiaomi 2.1.1_0159 or ATC_ota_40000 again also gives the same result. Does anyone know a solution to bring this sensor back to life?

VicCoca, 2025-04-27:
There's a NB saying this version is currently not supported. I flashed it to the ATC version following [This video](https://www.youtube.com/watch?v=NfZHh6wmTp8) but now I'm not sure where to go. I wanted to use Zigbee since I have a zigbee hub, but not a bluetooth one.