devbis/z03mmc #180: Values on display don't reported values

ezcGman, 2025-01-22:
Hey there,

I have a weird issue where the temperature values shown on the display are in 9/10 cases not matching the values reported via Zigbee, even waiting 30s (default display refresh timeout) doesn't fix it. Is that know or did I miss a setting somewhere?
I did not do any calibration, I just flashed the latest firmware and that's it.

Thanks and greetings,

Andy!

ezcGman, 2025-01-22:
"Unknown":
[image]

I've flashed the latest from this repo, 1.1.0

reinharty, 2025-04-28:
I also observe this behavior with my newly flashed LYWSD03MMC using 1.1.0. In ZHA the firmware is displayed as "unknown".
No manual configuration, like calibration, of the device happened. I tried looking for values wich set the refresh intervals for display and Zigbee but was unable to find ist.

giuseppe99barchetta, 2025-05-30:
Same problem here

gresnyer, 2025-10-23:
I have the same issue. On the display always showing 0.6 C offset temperature. Humidity value displayed correctly. I am using latest fw (1.1.0) and z2m.