devbis/z03mmc #182: LYWSD03MMC bricked after flashing z03mmc.bin – no display, no Bluetooth detection

MVinualez, 2025-02-19:
Hi,

I flashed the `z03mmc.bin` firmware on my `LYWSD03MMC` thermometer, but since then:

The screen won’t turn on.
The device is no longer detected via Bluetooth, preventing me from reflashing it.
I’ve tried removing and reinserting the battery, but it didn’t help. Is there any way to recover the device (recovery mode, test points, another firmware, etc.)?

Thanks in advance for your help!

devbis, 2025-02-20:
https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-firmware-with-usb-to-uart

MVinualez, 2025-02-20:
> https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-firmware-with-usb-to-uart

Thanks for your suggestion! I looked into the UART method, but it requires a TTL UART adapter that costs around €10, which is the price of two new Xiaomi thermometers. In that case, I’m wondering if it’s really worth investing in the adapter just to attempt a recovery.

Is there any other possible solution, or could the Bluetooth recovery mode work in my case?

MVinualez, 2025-02-20:
Maybe there’s another procedure I could try? The battery voltage is present on the board when I insert it, but I still have no display or Bluetooth detection.
I also tried shorting GND and RESET for 5 seconds, but nothing happened. Do you have any other suggestions?

MVinualez, 2025-02-20:
I think I found the issue, I forgot to flash the intermediate ATC_ota_40000.bin and flashed z03mmc.bin directly.

Do you know if there’s any way to recover the device in this case?

devbis, 2025-02-20:
> TTL UART adapter that costs around €10 

There are plenty of offers on aliexpress for <$1 . But I don't know other ways to recover it if the device doesn't boot.

MVinualez, 2025-02-22:
Okay thanks, I'll take a look!

closed 2025-02-22 by MVinualez