devbis/z03mmc #187: Can't update OTA in Z2M

fryc88, 2025-06-04:
Hi,
I'm looking for advice how to update it OTA through Z2M.
I have tried their repo and it shows me no update available. I have tried adding file to local folder as in the past and it still shows me no update available, even if i add 'force' line.
Any clues guys?

[image]

And i want to update it to 3001-1100 so i have added file and pointed it in ota json:

[image]

And all i've got is no update available

[image]

sendorm, 2025-06-06:
I was on the same train and was able to update like this: 
put this file under ota folder: 1141-0203-11003001-z03mmc.zigbee
create a my_index.json file under the same place as your configuration.yaml with content: 
```
[
  {
    "url": "ota/1141-0203-11003001-z03mmc.zigbee",
    "force": true
  }
]
```
in your configuration.yaml file add this:
```
ota:
  zigbee_ota_override_index_location: my_index.json
```
after this restart zigbee and check for update. it found the update and updated successfully afterwards.

fryc88, 2025-06-06:
Just looked into log but it is not showing if it is looking for file or anything else, just got info about current and latest versions being same

[06-06-2025 19:40:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/kitchen_xiaomi', payload '{"battery":83.5,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"enable_display":null,"humidity":59.19,"humidity_calibration":null,"last_seen":"2025-06-06T19:40:34+01:00","linkquality":255,"show_smiley":null,"temperature":19.54,"temperature_calibration":null,"temperature_display_mode":"fahrenheit","update":{"installed_version":285224961,"latest_version":285224961,"state":"idle"},"update_available":false,"voltage":2400}'