devbis/z03mmc #189: Hardware B1.6 - blank LCD

deveon95, 2025-07-04:
I've flashed this unit using the UART method.

The flashing was successful and it was seen by Zigbee2MQTT but the screen is blank. Toggling the screen on setting doesn't turn it on,

Per the instruction "If you flashed the module but the screen is remaining blank, try python3 TLSR825xComFlasher.py -p <YOUR_COM_PORT> ea to erase all flash and then write the firmware again", I erased the flash and programmed it again but the screen remains blank.

By "blank", I mean the screen looked like its powered but with all the segments off, because all the segments look faint if held at an angle. The segments don't ever turn on, not even during power-up.

Is this hardware revision supported by the firmware?

[image]
[image]

devbis, 2025-07-07:
All know revisions are supported including this one. It has the same schematics as B1.5.
Try to erase and upload more times, as UART dongle/script is just an emulator for the SWS protocol that is used.

francescovannini, 2026-01-06:
Just for additional context, I have some B1.4 (running devbis fw) and recently I've added some 2025 B1.6 (running pvvx fw) and they all work perfectly, reporting the exact same values, tested with sensors sitting close to each other for a few weeks in a room that gets heated and cooled regularly. Battery consumption is also comparable; I don't know if I have clones or not, but for me these are as good as the original ones.

devbis, 2026-01-07:
@francescovannini Can you please try the updated version of the firmware with SPI display support included?
Unfortunately, I don't have such version so I can't test is myself. 
*.zigbee file for local OTA and *.bin file for flashing over wires. 

[firmware_20260107.zip](https://github.com/user-attachments/files/24467682/firmware_20260107.zip)

oguzatagan, 2026-10-31:
hi
is there a improvoment? i install this firmware zigbee working but screen is off